Kinovarobotics/kinova-ros #168: Single threaded spinner after multi-threaded ones in kinova driver

Achllle, 2018-09-28:
Both the [joint_trajectory_action_server](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp) and [gripper_command_action_server](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/joint_trajectory_action/gripper_command_action_server.cpp) launch a single-threaded spinner after launching an asynchronous spinner. This results in the warning/error below (from rqt_console):

```
Node: /m1n6s300_gripper_command_action_server
Time: 13:29:30.428035654 (2018-09-28)
Severity: Error
Published Topics: /m1n6s300_driver/fingers_action/finger_positions/cancel, /m1n6s300_driver/fingers_action/finger_positions/goal, /m1n6s300_gripper/gripper_command/feedback, /m1n6s300_gripper/gripper_command/result, /m1n6s300_gripper/gripper_command/status, /rosout

SpinnerMonitor: single-threaded spinner after multi-threaded one(s).Attempt to spin a callback queue from two spinners, one of them being single-threaded.
In the future this will throw an exception! Only allowed for backwards compatibility.
```

martine1406, 2018-10-03:
Hi @Achllle 

Thanks for letting us know. Can you confirm exactly what commands are you running so I can try to replicate the error. Maybe this commit from the TurtleBot arm project could help solve this non-fatal error: https://github.com/turtlebot/turtlebot_arm/commit/afacc4cfa58330cfc32285b43e6f1141054a538e

Achllle, 2018-10-04:
I'm running:
```
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s300
roslaunch m1n6s300_moveit_config m1n6s300_demo.launch
```
I'll look into that commit

martine1406, 2018-10-04:
If I understand correctly, this error does not currently affect the functionality of the node, but might be affecting it in the future?

Achllle, 2018-10-04:
Correct, a non-fatal error. An error nevertheless and we'd like to not have it show up on launch each time.

martine1406, 2018-10-09:
Hi @Achllle 

I am trying to reproduce your issue, but I do not get the same message in rqt_console. I just have one type of message with severity warn, and it is 
Node: /move_group
Time: 17:32:04.540871851 (2018-10-09)
Severity: Warn
Published Topics: /rosout

Skipping virtual joint 'world_frame_fixed_joint' because its child frame 'root' does not match the URDF frame 'world'

Location:
/tmp/binarydeb/ros-indigo-moveit-core-0.7.13/robot_model/src/robot_model.cpp::903

----------------------------------------------------------------------------------------------------
I have no message with severity error. I am unsure what is the difference between both our setups. I launch/run exactly the same commands as you 

first I run rqt_console:
`rosrun rqt_console rqt_controle`

then
```
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s300
roslaunch m1n6s300_moveit_config m1n6s300_demo.launch
```

If you want to share a video of your screen and what you see in rqt_console, it might be helpful. I else I do not know if you just have a different version of roscpp than me (I think this error message comes from roscpp)

Cheers
Martine

Achllle, 2018-10-10:
I just realized that this is most likely due to you using an older version of ROS; Indigo. In that version using two different threading models in one node is not deprecated yet. It's deprecated in Kinetic and probably throws an error in Melodic.

martine1406, 2018-10-10:
I see. Are you testing on Kinetic then?

trkygt, 2019-02-22:
Is there any update about the issue that been addressed by AchIlle?

nsaif-kinova, 2019-02-22:
Hello Tarik,
Which version of ROS are you using?
https://github.com/Kinovarobotics/kinova-ros/issues/168#issuecomment-428393777

trkygt, 2019-02-22:
Hello Nassif,
Thank you for the prompt response. We have two different PCs with kinetic and melodic in our lab. 

I just fixed the issue thanks to the commit [24dc404](https://github.com/CNURobotics/kinova-ros/commit/24dc404d542e67451a032b893e0f209b44c103ef) on melodic version.

But, I do have another error message when I run demo.launch. I will look into that and if I can't fix it and I will open a new issue for that. Thanks again!

nsaif-kinova, 2019-02-22:
Thank you Tarik for the fix,
Please note Kinova officially support ROS indigo on Jaco, our Kinetic version was developed by the community and not officially tested by Kinova.
Sincerely,

trkygt, 2019-02-22:
Yes, I am aware of that. Thank you for the clarification. I think this issue can be closed now?

closed 2019-02-22 by nsaif-kinova