Kinovarobotics/kinova-ros #169: MoveIt trajectories do not complete on the Mico2 arm

pirobot, 2018-10-02:
We have a Mico2 arm with 2-finger gripper and we are running ROS Kinetic on Ubuntu 16.04.  When we set a goal pose for the gripper in MoveIt, the arm begins moving correctly along a trajectory that looks like it will get the gripper to the correct pose.  However, the trajectory never completes on the first try.  Instead, it gets about 90% of the way there and then stops.  If we re-plan and re-execute, then the arm continues the rest of the way.  We have tried increasing the following parameters with no effect: 

- trajectory_execution/allowed_execution_duration_scaling = 4.0
- move_group/trajectory_execution/execution_duration_monitoring = false
- trajectory_execution/allowed_goal_duration_margin = 2.0
- trajectory_execution/allowed_start_tolerance = 0.01

Here are two examples showing where the arm ended up on the first execution attempt and then where it got on the second attempt.

[image]
[image]
[image]
[image]

nsaif-kinova, 2018-10-03:
Hello pirobot,
Can you please share your initial and final positions? I will try to replicate the problem on my side and see why this is happening.
Sincerely,

pirobot, 2018-10-04:
Thanks @nsaif-kinova.  Here are the joint position values for the two poses shown above.  These are named poses in my MoveIt config file ("wave" and "tuck") and I am just trying to go back and forth between them:

wave: 4.716292109278309, 1.0713765238914816, 2.3086940826735827, -3.718741008426185e-05, -0.001899037105311643, -0.0012643720078834842

tuck: 4.713287530936483, 4.438145706076036, 5.6696409289249505, -2.4168075039703303, 0.8285355141023337, -0.6911503571581435

pirobot, 2018-10-04:
I just discovered that if I run the kinova_demo joints_action_client.py script with this same joint position targets, it works perfectly.  In other words, if I run:

$ rosrun kinova_demo joints_action_client.py m1n6s200 radian 4.716292109278309 1.0713765238914816 2.3086940826735827 0.0 -0.001899037105311643 -0.0012643720078834842

Then the arm goes all the way to the final "wave" position without stopping before it gets there.  Same thing for the "tuck" pose.  So it seems to be a problem with the MoveIt wrapper or its parameters...

martine1406, 2018-10-04:
@pirobot 

When running on moveit, are you only running in simulation or with the real robot? If running with the real robot, please see this related issue I think #115  https://github.com/Kinovarobotics/kinova-ros/issues/115. If only running in simulation, that might be different. 

When you are using 

`$ rosrun kinova_demo joints_action_client.py m1n6s200 radian 4.716292109278309 1.0713765238914816 2.3086940826735827 0.0 -0.001899037105311643 -0.0012643720078834842, `

you are actually sending an angular position directly to the real robot. Moveit is not involved in that whatsoever. 

Let me know about my first question
Martine

pirobot, 2018-10-04:
@martine1406 

I am doing all of this on the real robot, so I will take a look at issue #115.

martine1406, 2018-10-30:
@pirobot

Has this issue been solved?

pirobot, 2018-11-01:
@martine1406 Sorry we have not had a chance to look at this again but will eventually.

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve