Kinovarobotics/kinova-ros #170: Cannot Switch to Torque Control Mode When the Robot is Holding a Payload

sahandrez, 2018-10-05:
Hi, 

I am using the torque control mode on the Jaco 2 arm. So far, I was able to successfully control it when the robot is not holding anything in its hand. Now, I wanted to try the torque controller when the robot has a 0.9 kg payload and it is not switching to the torque control mode successfully, even though the log in the terminal says it has switched to torque control mode. 

I have set the safety factor to 1.0. 

Is this some kind of a safety feature? For example, since the initial torques are larger than normal due to the payload, the robot base refuses to switch to torque control mode?

Thank you very much. 

Bests, 
Sahand

nsaif-kinova, 2018-10-05:
Hi,
The robot is sensing a big force on the end effector and will not let you switch to torque mode for your safety, if the robot switch to torque while holding a payload will follow the gravity of the payload and accelerate.
To solve this problem you should tell the robot that there is a payload attached, to do that you can use the Torque Console and specify a payload, or you can adjust your general parameters which is by default payload: [0, 0, 0, 0] payload: [COM COMx COMy COMz] in [kg m m m]
Sincerely,

sahandrez, 2018-10-05:
Hi, 

Thank you for your reply. 
What are the COMx, COMy and COMz exactly? Are they the coordinates of the COM of the payload with respect to the last joint? 

Bests, 
Sahand

nsaif-kinova, 2018-10-05:
These are the payload’s center of mass with respect to Kinova robot’s effector reference frame.

sahandrez, 2018-10-05:
It's working now! Thank you.

closed 2018-10-05 by sahandrez

sjhansen3, 2018-10-22:
Is there a way to turn this safety feature off or down? We are using torque mode when we stand next to the robot, if the robot drops to the ground that is fine, we are holding it in our hands.

martine1406, 2018-10-22:
Hi @sjhansen3 

No unfortunately, your only work around is to send that payload to the robot is it can take it into account in its gravity estimation.