Kinovarobotics/kinova-ros #172: No movement on real robot

GianlucaCerilli, 2018-10-15:
Hi,

I have a j2n6s300 Jaco robot and I would like to move it to a certain goal pose.
I followed these steps (with the robot in home position):

- **roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300**
- **roslaunch robot_name_moveit_config robot_name_demo.launch**
- selected _current_ (start state) and _update_ in rviz
- selected _vertical_ (goal state) and _update_ in rviz
- selected _plan_ (and I see the grey manipulator moving)
- selected _execute_ 
At this step I don't see any movement, neither in the simulator nor in the real robot, but on the console it is written:

> [ INFO] [1539620921.508211080]: Received new trajectory execution service request...
> [ INFO] [1539620921.597989112]: Joint_trajectory_action_server received goal!
> [ INFO] [1539620921.598172281]: Joint_trajectory_action_server accepted goal!
> [ INFO] [1539620921.598330803]: Joint_trajectory_action_server published goal via command publisher!
> [ INFO] [1539620928.396945425]: Completed trajectory execution with status SUCCEEDED ...
> [ INFO] [1539620928.397287469]: Execution completed: SUCCEEDED

Can you tell me why the real robot is not moving?
Thank you very much

nsaif-kinova, 2018-10-15:
Hello GianlucaCerilli,
I tried the same steps on my robot and I am able to plan and execute a a goal in MoveIt, the robot is also following the same movement.
Are you using ROS Indigo?
Sincerely,

GianlucaCerilli, 2018-10-15:
Hi nsaif-kinova,
Thank you for your reply. I am using Ubuntu 16.04 with Ros Kinetic. But moveit and all the rest is correctly working with it.. So I don't know what could be the problem.

martine1406, 2018-10-15:
do you have any output in your terminal where you launch roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300

GianlucaCerilli, 2018-10-15:
Hi Martine,

I have no output in that terminal after running moveit with rviz

GianlucaCerilli, 2018-10-16:
This is the terminal for the **roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300** :

> ... logging to /home/gianluca/.ros/log/d9ad25ba-d114-11e8-86a4-94b86df88095/roslaunch-gianluca-Precision-5530-11954.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> xacro: Traditional processing is deprecated. Switch to --inorder processing!
> To check for compatibility of your document, use option --check-order.
> For more infos, see http://wiki.ros.org/xacro#Processing_Order
> xacro.py is deprecated; please use xacro instead
> started roslaunch server http://gianluca-Precision-5530:40515/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /j2n6s300_driver/connection_type: USB
>  * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
>  * /j2n6s300_driver/ethernet/local_cmd_port: 25000
>  * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
>  * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
>  * /j2n6s300_driver/jointSpeedLimitParameter1: 10
>  * /j2n6s300_driver/jointSpeedLimitParameter2: 20
>  * /j2n6s300_driver/robot_name: j2n6s300
>  * /j2n6s300_driver/robot_type: j2n6s300
>  * /j2n6s300_driver/serial_number: not_set
>  * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
>  * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
>  * /kinova_number_of_robots: 2
>  * /kinova_robots: [{'serial': 'PJ00...
>  * /robot_description: <?xml version="1....
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
>   /
>     j2n6s300_driver (kinova_driver/kinova_arm_driver)
>     j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[j2n6s300_driver-1]: started with pid [11974]
> process[j2n6s300_state_publisher-2]: started with pid [11975]
> [ INFO] [1539675088.046817561]: kinova_robotType is j2n6s300.
> [ INFO] [1539675088.047142567]: kinova_robotName is j2n6s300.
> [ INFO] [1539675088.049821112]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
> [ INFO] [1539675088.206368194]: Found 1 device(s), using device at index 0 (model: Jaco V2 Service, serial number: PJ00900006161110001, code version: 393494, code revision: 6)
> [ INFO] [1539675088.280301818]: Initializing fingers...this will take a few seconds and the fingers should open completely
> [ INFO] [1539675088.431798079]: The arm is ready to use.
> j2n6s300_joint_1 j2n6s300_joint_2 j2n6s300_joint_3 j2n6s300_joint_4 j2n6s300_joint_5 j2n6s300_joint_6 

And this is the terminal for the **roslaunch j2n6s300_moveit_config j2n6s300_demo.launch** :

> ... logging to /home/gianluca/.ros/log/d9ad25ba-d114-11e8-86a4-94b86df88095/roslaunch-gianluca-Precision-5530-12246.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> xacro: Traditional processing is deprecated. Switch to --inorder processing!
> To check for compatibility of your document, use option --check-order.
> For more infos, see http://wiki.ros.org/xacro#Processing_Order
> xacro.py is deprecated; please use xacro instead
> started roslaunch server http://gianluca-Precision-5530:45977/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /controller_list: [{'default': True...
[... 262 lines cut ...]
> [ INFO] [1539675221.564624888]: RRTConnect: Starting planning with 1 states already in datastructure
> [ INFO] [1539675221.564708631]: RRTConnect: Starting planning with 1 states already in datastructure
> [ INFO] [1539675221.565510815]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.565770428]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.566061960]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.566183781]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.566285197]: ParallelPlan::solve(): Solution found by one or more threads in 0.001830 seconds
> [ INFO] [1539675221.566371800]: RRTConnect: Starting planning with 1 states already in datastructure
> [ INFO] [1539675221.566423286]: RRTConnect: Starting planning with 1 states already in datastructure
> [ INFO] [1539675221.567235312]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.567258788]: RRTConnect: Created 5 states (2 start + 3 goal)
> [ INFO] [1539675221.567394125]: ParallelPlan::solve(): Solution found by one or more threads in 0.001052 seconds
> [ INFO] [1539675221.570944656]: SimpleSetup: Path simplification took 0.003489 seconds and changed from 3 to 2 states
> [ INFO] [1539675221.571582488]: Planning adapters have added states at index positions: [ 0 ]
> [ INFO] [1539675225.505017409]: Received new trajectory execution service request...
> [ INFO] [1539675225.548000573]: Joint_trajectory_action_server received goal!
> [ INFO] [1539675225.548230797]: Joint_trajectory_action_server accepted goal!
> [ INFO] [1539675225.548400106]: Joint_trajectory_action_server published goal via command publisher!
> [ INFO] [1539675234.346898869]: Completed trajectory execution with status SUCCEEDED ...
> [ INFO] [1539675234.347204269]: Execution completed: SUCCEEDED

Only this second terminal reports some output after the _plan_ and _execute_ commands, but neither in the simulation nor in the real robot I can see the robot moving after the _execute_ command.

I really don't know what to do.
Thank you very much

martine1406, 2018-10-16:
Hi @GianlucaCerilli 

The output from the first terminal indicates that the connection with the real robot is successful. 
Can you send a printscreen of what you are seeing in Rviz when launching roslaunch j2n6s300_moveit_config j2n6s300_demo.launch?

As Nassif said, we officially support ROS Indigo. The two exact same steps you do (roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300, then roslaunch j2n6s300_moveit_config j2n6s300_demo.launch) work fine on my Ubuntu 14.04 computer with ROS indigo. 

Thank you
Martine

GianlucaCerilli, 2018-10-16:
Hello Martine,

I did not know that you supported only Indigo...since you are more experienced maybe you can find the problem, but I don't want to bother you. By the way, this is the screen when Rviz opens:

[image]

And this is the other screen when I update the goal pose before doing execute:

[image]

Thank you very much

martine1406, 2018-10-16:
Hi @GianlucaCerilli 

Is you real robot initially in the same position (home position) as shown in the first picture?

GianlucaCerilli, 2018-10-16:
Hi Martine,
Yes, I put it there with **rosservice call /j2n6s300_driver/in/home_arm**. In fact, the grey pose in the second image keeps updated with the movements of the real robot. But the contrary does not happen.
As you can see in the second terminal it tells me that I have unnormalized quaternion. Indeed, If I use **rostopic echo /j2n6s300_driver/out/tool_pose** I get:
[image]
I think this is the pose of the end effector wrt to the base (p.s. how can I get the poses of the other links, included the link_base?). 
I don't know if it is normal to have it in this format, but I don't think this is related to the fact that the robot does not move...

GianlucaCerilli, 2018-10-17:
Hi @martine1406 

Yes, when I click _plan_ I can see the grey robot moving to the desired position. It is only the _execute_ command that does not working in both systems (even if I get successful results on the terminal).
Thank you very much Martine1406 for your availability. Let me know when you get some results.

> If you want the other frames in ROS, you have to monitor the tf transforms (with tf echo). The output you get seems appropriate (anything you feel is wrong?)

In the output from /j2n6s300_driver/out/tool_pose (and also if I look at the other link poses in rviz) I have the quaternions that are not normalized, so I was wondering if this is something that I should correct. 
In addition to this, If I use **rostopic echo /tf** for the _link_base - link_1_ frames, I get:
[image]
where the translation on Z axis is about 0.15675, while the "base to shoulder" D1 distance should be 0.2755.

So I am not sure if the parameters that I get from _echo /tf_ and _/out/tool_pose_ are correct or if it is necessary to correct something (e.g. normalizing, ...). 

For the other frames, if I use rostopic echo /tf I get the transforms from one frame to the other, but not directly the pose of each frame, I guess. Is it possible do it? For example, getting the link-base frame pose (that I guess is position: 0,0,0 orientation: 0,0,0,1).

Thank you very much!
Sincerely

martine1406, 2018-10-17:
Hi @GianlucaCerilli 

You can certainly have the same end effector frame in Rviz and on the real robot (that would actually make more sense, I'll check if we can correct that in the package). If you want to correct in on your end, you just have to modify the urdf file and change the joint_end_effector_origin_rpy. I do not specifically think that you need to have equivalent intermediate frame between the real robot and Rviz, but let me know if you think otherwise. 

Martine

GianlucaCerilli, 2018-10-18:
Hi @martine1406 ,

I think it could be better to have the possibility of getting all the real frames' poses for real applications and not (or not only, at least) the Rviz frame poses. 
I also wanted to ask two additional informations if possible.
Is there any repository for this j2n6s300 for calibrating a camera mounted on the end effector? 
And, since the robot is mounted on the provided kit, a translation on Z axis for the base frame (as world frame) should be taken in consideration. Is it reported somewhere the precise measure of this translation?

Thank you again and sorry for the many questions. Let me know when you will get some result or correction.

Gianluca

martine1406, 2018-10-18:
Hi @GianlucaCerilli 

Rviz need the frames positioned at the joints. The robot does not have all the intermediate frames positioned at the joints. What you can do is perhaps define additional tf frames that will fit the real robot's frames and allow you to use tf echo to track those frames' motion without impairing the display in Rviz?

For you two additional questions:

- Is there any repository for this j2n6s300 for calibrating a camera mounted on the end effector? 
no, no ready-to-use repo for this kind of process :-( Although we might be able to advise if you have specific questions. 

- And, since the robot is mounted on the provided kit, a translation on Z axis for the base frame (as world frame) should be taken in consideration. Is it reported somewhere the precise measure of this translation? 
no. The reason for that is that many people decide to mount the robot on custom-made mountings made from standard aluminum extrusions instead of the provided kit. Can you measure this translation on your mounting if that's not too complicated?

Thank you

martine1406, 2018-10-18:
We will correct the package in the next few days. I have someone working on it. 
I am be able to test with a robot on Ubuntu 16.04 at the end of the day. I have some priorities I have to handle first, but thanks for the reminder :)

martine1406, 2018-10-22:
HI @GianlucaCerilli 

The urdf is corrected. However, we tested on Ubuntu 16.04 and could not reproduce you isue with the execute.

We proceeded the following way
- clone the package on the Ubuntu 16.04
- checkout to commit 94c25e1
- roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300
- roslaunch j2n6s300_moveit_config j2n6s300_demo.launch

I am not sure how we can help more :-O

martine1406, 2018-10-22:
Hi again

Finally, the person who did the test had tested on the wrong computer. When testing on Ubuntu 16.04, we also have a problem at the execution. However, in my case, I see the goal being aborted, not succeeded, in the moveit terminal. According to the error message, it seems that moveit has trouble accessible the robot's joint_state within 1 ms. As this package is not officially supported on Ubuntu 16.04, I cannot really spend much time right now on the investigation, but still, I'll let you know if I have something. Sorry about that.
Martine

GianlucaCerilli, 2018-10-24:
Hi @martine1406 ,

Thank you very much! Let me know If you will figure out the problem.

Thank you also for the urdf correction. I will update the moveit.rviz file according to your changes.
Can you please tell me what are the values of the real end-effector frame that you get, for example at home position? So, I can compare with mine to understand if the values that I get are the correct ones.

Thank you again!
Gianluca

martine1406, 2018-10-24:
from rostopic echo /j2n6s300_driver/out/tool_pose
pose: 
  position: 
    x: 0.212557017803
    y: -0.256574690342
    z: 0.508452177048
  orientation: 
    x: 0.643426954746
    y: 0.320419073105
    z: 0.422684729099
    w: 0.55197006464

from rosrun tf tf_echo /j2n6s300_link_base /j2n6s300_end_effector
- Translation: [0.213, -0.257, 0.508]
- Rotation: in Quaternion [0.643, 0.320, 0.423, 0.552]
            in RPY (radian) [1.605, -0.191, 1.109]
            in RPY (degree) [91.947, -10.970, 63.546]

GianlucaCerilli, 2018-10-25:
Thank you @martine1406,

The problem is that I am on the commit 94c25e1 (from branch kinova-ros-beta) that works on kinetic but you did the last commit with the changes for the urdf on the branch master, that does not compile on kinetic. 
Is it possible to update that changes also from my position or from branch kinova-ros-beta?

martine1406, 2018-10-26:
just replace the needed urdf files in your repo after you switched to commit 94c25e1? The changes for this bug are in all urdf files (robot_code).xacro (e.g. j2n6s300.xacro)

GianlucaCerilli, 2018-10-26:
Thank you very much @martine1406 ,
I have this result now, a bit different from yours, but I definitively think is the correct one:

rostopic echo /j2n6s300_driver/out/tool_pose
> pose: 
>   position: 
>     x: 0.213109344244
>     y: -0.256380349398
>     z: 0.509006023407
>   orientation: 
>     x: 0.642293870449
>     y: 0.319399952888
>     z: 0.425574630499
>     w: 0.551659822464

rosrun tf tf_echo /j2n6s300_link_base /j2n6s300_end_effector
> - Translation: [0.213, -0.256, 0.509]
> - Rotation: in Quaternion [0.642, 0.319, 0.426, 0.552]
>             in RPY (radian) [1.600, -0.196, 1.113]
>             in RPY (degree) [91.701, -11.203, 63.758]

Thank you again!
Gianluca

martine1406, 2018-10-26:
great! Yes it might depend slightly on the encoders' readings.

woshisj, 2018-12-28:
> great! Yes it might depend slightly on the encoders' readings.
Hi I checkout to commit 94c25e1  in ubuntu16
and the real robot not movement in rviz
This problem solved？

woshisj, 2018-12-28:
> just replace the needed urdf files in your repo after you switched to commit [94c25e1](https://github.com/Kinovarobotics/kinova-ros/commit/94c25e19443acb4fadac34c3f796f108f6a8471b)? The changes for this bug are in all urdf files (robot_code).xacro (e.g. j2n6s300.xacro)

Hi I checkout to commit 94c25e1 in ubuntu16
and the real robot not movement in rviz
This problem solved？

g1y5x3, 2019-07-24:
I am having the same issue. I was looking at the topic connection from rqt_graph and noticed that there is a topic **/m1n6s200_driver/trajectory_controller/command** is not subscribed by the **m1n6s200_driver** node.
[image]

@martine1406 Can you possibly show me the rqt_graph on your working version? I just want to see the exact connection through different topics between all the nodes.

Thanks so much!

g1y5x3, 2019-07-25:
> I am having the same issue. I was looking at the topic connection from rqt_graph and noticed that there is a topic **/m1n6s200_driver/trajectory_controller/command** is not subscribed by the [...]

I have found a solution. Here is the link to my answer! [mico cannot move using moveit #234](https://github.com/Kinovarobotics/kinova-ros/issues/234#issuecomment-515147522)

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve