Kinovarobotics/kinova-ros #173: SRDF unable to load in Movit

2016mez0019, 2018-10-18:
I am trying to run the simulation of kinova mico robot in Moveit but I am getting this error at the start:
after getting error in moveit, I actually checked srdf file, I got this error message: 

anubhav@anubhav-VirtualBox:~/catkin_ws/src/kinova-ros-master/kinova_moveit/robot_configs/m1n6s300_moveit_config/config$ check_urdf m1n6s300.srdf 
Error:   No link elements found in urdf file
         at line 180 in /build/buildd/urdfdom-0.2.10+dfsg/urdf_parser/src/model.cpp
ERROR: Model Parsing the xml failed

martine1406, 2018-10-18:
Hi @2016mez0019 

Please provide the exact commands you are writing in the terminal and tell us which version of ROS you are using. 

Thank you

martine1406, 2018-10-30:
any news? can I close this issue?

closed 2018-11-02 by martine1406