Kinovarobotics/kinova-ros #176: Finger Action Server not working

Behery, 2018-10-26:
Hi, 
I'm trying to move the fingers of my JACO 2 arm, using the demo finger position client but it's not working.
I'm using the demo from the readme:
```
$ rosrun kinova_demo fingers_action_client.py j2n6s300 percent -- 100 100 100
   obtained current finger position 
   Sending finger position ...
   Finger position sent!
```
But on the bring-up terminal I'm getting the following:
```
[DEBUG] [1540566817.171801810]: The action server has received a new goal request
[DEBUG] [1540566817.171956712]: A new goal has been recieved by the single goal action server
[DEBUG] [1540566817.172216194]: Accepting a new goal
[DEBUG] [1540566817.172275810]: Accepting goal, id: /j2n6s300_gripper_workout-1-1540566817.141, stamp: 1540566817.14
[ERROR] [1540566817.184697382]: KinovaCommException: Wrong control type (return code: 1)

[DEBUG] [1540566817.184956713]: Setting the current goal as aborted
[DEBUG] [1540566817.185078694]: Setting status to aborted on goal, id: /j2n6s300_gripper_workout-1-1540566817.141, stamp: 1540566817.14
[DEBUG] [1540566817.185192306]: Publishing result for goal with id: /j2n6s300_gripper_workout-1-1540566817.141 and stamp: 1540566817.14
[DEBUG] [1540566817.188516965]: void kinova::KinovaFingersActionServer::actionCallback(const SetFingersPositionGoalConstPtr&): LINE 177, setAborted 
```

In the result topic I'm getting:
```
$ rostopic echo /j2n6s300_driver/fingers_action/finger_positions/result 
header: 
  seq: 0
  stamp: 
    secs: 1540566817
    nsecs: 185163019
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1540566817
      nsecs: 140717983
    id: "/j2n6s300_gripper_workout-1-1540566817.141"
  status: 4
  text: ''
result: 
  fingers: 
    finger1: 0.25
    finger2: 0.25
    finger3: 0.25
---
```
What am I doing wrong here? how can I set the right control type?

Thanks :)

nsaif-kinova, 2018-10-26:
Hello Anwar,
When you try to move the robot fingers using a joystick are they moving?
Please note when you turn the robot ON you have to send it Home and initialise the fingers by opening them completely, otherwise you will not be able to use them.
 
Usually `roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300` will open the fingers completely to initialise them. Is this happening when you launch the command?

`rosrun kinova_demo fingers_action_client.py j2n6s300 percent --100 100 100` will completly close the fingers.
`rosrun kinova_demo fingers_action_client.py j2n6s300 percent -- 0 0 0 `will completly open the fingers.

To monitor the finger positions please use this command `rostopic echo /j2n6s300_driver/out/finger_position` 0 means completely open, a value over 6000 means fingers closed.
 
Please let me know if this will help solving your problem, sincerely,

Behery, 2018-10-26:
Hi @nsaif-kinova 
Thanks for the fast reply!! :) 
I'm able to move the arm using a joystick and a spacemouse (using the /j2n6s300_driver/in/cartesian_velocity topic). When the bringup node starts the fingers open fully, only afterwards I can move the arm through ros (but not the fingers). I can move the fingers using the joystick.

The /out/finger_position topic shows the following when the fingers are fully open:
```
$ rostopic echo /j2n6s300_driver/out/finger_position
finger1: 0.25
finger2: 0.25
finger3: 0.25
---
finger1: 0.25
finger2: 0.25
finger3: 0.25
---
```

martine1406, 2018-10-26:
Hi @Behery 

Can you try to send the arm to launch the bringup, send the arm to the home position, then send the command to the fingers through ROS? From the error you are getting, it seems because the arm is in a kind of no-man's-land when you just started it (it is neither in Cartesian nor in angular mode). Sending it to the home position will automatically activate Cartesian mode. Let us know.

Behery, 2018-10-26:
Hi Martin,

not sure what you mean by
> try to send the arm to launch the bringup 

But I home the arm using the button on the joystick then I start the bringup. If I start the bring up then home it (or move it) using the joy stick, then I'm no longer able to move it through ros.
Is there a way to start hte cartesian control using ros?

martine1406, 2018-10-26:
Hi 

Yes I must admit it is unclear! Sorry, I started writing this sentence and changed my mind in the middle. 
I mean roslaunch kinova_bringup. 

> But I home the arm using the button on the joystick then I start the bringup. If I start the bring up then home it (or move it) using the joy stick, then I'm no longer able to move it through ros.

This is not normal. You should be able to do both. You should be able to home the arm using the joystick, then when the robot is at home, start the bringup and control the robot in ROS, and you should also be able to start de bringup, then home it using the joystick and when the robot is at home, control the robot in ROS. Can you send any error output you get when trying to control the arm in ROS after homing it.

You can always decide to send the arm to its home position via the rosservice j2n6s300_driver/in/home_arm, which will set the arm in Cartesian mode once the home position is reached. 

Let me know

Behery, 2018-10-27:
If I use the joystick (to home or move the arm) then ros, ros works and moves normally, It used to not work before for some reason and I thought it was normal. Except if I retract the arm, then home it, after that I can't move it.

The `/j2n6s300_driver/in/home_arm` isn't working, it's neither outputting an error nor moving the arm
```
$ rosservice call /j2n6s300_driver/in/home_arm "{}" 
homearm_result: "KINOVA ARM HAS BEEN RETURNED HOME"
```
In the bring up:
```
[ INFO] [1540642198.145117588]: Homing the arm
[ INFO] [1540642198.235154764]: Initializing fingers...this will take a few seconds and the fingers should open completely
```

If I home using the joystick (or call the home_arm service), then use the `fingers_action_client`, it still shows the same error as above:
```
[DEBUG] [1540642393.720957610]: The action server has received a new goal request
[DEBUG] [1540642393.721108341]: A new goal has been recieved by the single goal action server
[DEBUG] [1540642393.721292056]: Accepting a new goal
[DEBUG] [1540642393.721411730]: Accepting goal, id: /j2n6s300_gripper_workout-1-1540642393.717, stamp: 1540642393.72
[ERROR] [1540642393.728064198]: KinovaCommException: Wrong control type (return code: 1)

[DEBUG] [1540642393.728155146]: Setting the current goal as aborted
[DEBUG] [1540642393.728234828]: Setting status to aborted on goal, id: /j2n6s300_gripper_workout-1-1540642393.717, stamp: 1540642393.72
[DEBUG] [1540642393.728309671]: Publishing result for goal with id: /j2n6s300_gripper_workout-1-1540642393.717 and stamp: 1540642393.72
[DEBUG] [1540642393.728831091]: void kinova::KinovaFingersActionServer::actionCallback(const SetFingersPositionGoalConstPtr&): LINE 177, setAborted 
```

Behery, 2018-10-27:
> If I use the joystick (to home or move the arm) then ros, ros works and moves normally, It used to not work before for some reason and I thought it was normal.

This doesn't always happen, but I can't figure out the pattern why it sometimes works and sometimes not

martine1406, 2018-10-29:
Ok, let's start from the basics then. Let's forget ROS at the moment, and work with the Development Center available in the Kinova SDK. 

First of all, can you confirm on which version of Ubuntu you are working. Please remember that, officially, this ROS package is built for ROS Indigo/Ubuntu 14.04.

Can you also confirm your arm's serial number

Then, if you close anything ROS related, then open the Development Center. Can you send a printscreen of the General Settings->Update page. I want to see your base firmware and your actuators firmware. 

Finally, if you home the arm, then try to control the arm with the virtual joystick (including opening and closing the fingers), do you get any weird behaviour? Do you have any trouble controlling the robot. 

If we still cannot get a good hint on what this problem is, we might need to consider a video call. This is a behaviour that is totally unexpected for a Kinova arm. On our end, we have no problem controlling a robot with ROS after homing it, nor do we have problems calling to the home_arm service....

Thank you
Martine

Behery, 2018-10-29:
Hi Martine,
Thanks for your reply.

> Ok, let's start from the basics then. Let's forget ROS at the moment, and work with the Development Center available in the Kinova SDK.

The SDK crashes for some reason on Ubuntu during installation.

>First of all, can you confirm on which version of Ubuntu you are working. Please remember that, officially, this ROS package is built for ROS Indigo/Ubuntu 14.04.

I'm working on Kinetic/Ubuntu 16.04, which I'm guessing also explains why the sdk installation fails.

>Can you also confirm your arm's serial number

00000001030603123380002

> Then, if you close anything ROS related, then open the Development Center. Can you send a printscreen of the General Settings->Update page. I want to see your base firmware and your actuators firmware.

I installed the windows sdk but it's showing all the firmware versions as 0.0.0 and doesn't accept my serial number (which i guess it needs for communication with the arm). Same goes for the JACO soft application that came with the arm. Is there another way to read the firmware versions?

>If we still cannot get a good hint on what this problem is, we might need to consider a video call. 

That would be great to sort everything out.

martine1406, 2018-10-29:
HI @Behery 

The SDK is supposed to work on Ubuntu 16.04
The windows sdk needs usb drivers to communicate with the robot. Have you installed the drivers?

I have a colleague who could jump on a call with you this afternoon or tomorrow morning. When would you be available?

Also, I cannot find your arm's serial number. Normally, it is a number that starts with PJ something and that is written on a small label on the arm's base. My colleague will be able to confirm that information with you

thank you
Martine

Behery, 2018-10-29:
About the serial number, this is the one at the base of the robot on the sticker. It starts with PJ and then that number above.

martine1406, 2018-10-29:
Hi @Behery 

First of all, after asking a colleague, I could get the information using your arm's serial number and it seems this arm is NOT a JACO2, but a JACO1. This ROS package does not officially support JACO1 robot, so it might be the reason why you get all those errors and strange behaviours. 

Now that we know that, I will ask my colleague to test the finger control with a JACO1 and see if we can reproduce the problem internally. 
Thanks
Martine

martine1406, 2018-10-30:
this issue is most likely related to a Jaco1 firmware version that is incompatible with the current Kinova API. support@kinova.ca will handle this case.

closed 2018-10-30 by martine1406