Kinovarobotics/kinova-ros #177: Kinova moveit_core CMake Error when using ubunut 16.04

mambawockeez, 2018-10-31:
Im using ubuntu 16.04 to work on a project using kinova arms. and after cloning all the files, when i catkin_make it, it show error like this. And i already adding add_compile_options(-std=c++11) in every cmakelist.txt in kinova folder. And here is my error screenshot.
[image]

martine1406, 2018-10-31:
Please see this related issue #117

mambawockeez, 2018-10-31:
@martine1406 Hi, I reinstall moveit and try to cmake it and a new issue came up, https://github.com/ros-planning/moveit/issues/1178, could you plz check it and help me with that issue?

mambawockeez, 2018-10-31:
@martine1406 Im new to ros and kinova. So in this situation, after i cloning the git, what should i do to fix this before cmake it?

mambawockeez, 2018-10-31:
@martine1406 Im new to ros and kinova. So in this situation, after i cloning the git, what should i do to fix this before cmake it? Could you please be more specific
Sincerely

martine1406, 2018-10-31:
Hmm, I cannot really be more specific. You need to clone the package, checkout to commit 94c25e1.(https://git-scm.com/docs/git-checkout), install the required packages( requirements in https://github.com/Kinovarobotics/kinova-ros/wiki/Gazebo, https://github.com/Kinovarobotics/kinova-ros/wiki/MoveIt) and catkin_make. 

Are you a bit familiar with git? If not, there are certainly good references you can look at online.
Regards

mambawockeez, 2018-10-31:
@martine1406 Oh i see, i never use git checkout before. Thats why i dont understand the term 'checkout to commit'. So we use 'git checkout [-q] [-f] [-m] [--detach] <commit>' to checkout to commit 94c25e1 right?

mambawockeez, 2018-10-31:
@martine1406 Oh i see, i never use git checkout before. Thats why i dont understand the term 'checkout to commit'. So we use 'git checkout [-q] [-f] [-m] [--detach] <commit>' to checkout to commit 94c25e1 right? Could you please tell me what exactly should i type in the terminal? Im really appreciated your help

mambawockeez, 2018-11-01:
@martine1406 Got it! ill keep you update, recently im struggling with midterm not really have much time to continue this, but ill definitely let you know when i got time to fix this at the end of this week. really appreciated!

mambawockeez, 2018-11-08:
@martine1406 Everything works fine right now! thanks a lot! And could you please check my new issue?
https://github.com/Kinovarobotics/kinova-ros/issues/180

closed 2018-11-08 by mambawockeez