Kinovarobotics/kinova-ros #178: Could not initialize Kinova API (return code: 1010)

kurenaloha, 2018-11-01:
I try launch following, but I got the error.

roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300
-------------------------------------------------------------------------------
[ INFO] [1541044946.701735554]: kinova_robotType is j2n6s300.
[ INFO] [1541044946.709131005]: Initializing Kinova Ethernet API (header version: 50300, library version: 5.2.0)
[ERROR] [1541044946.709300482]: KinovaCommException: Could not initialize Kinova API (return code: 1010)

[right_driver-2] process has died [pid 24824, exit code -11, cmd /home/ytnpc2014b/jaco_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6s300 __name:=right_driver __log:=/home/ytnpc2014b/.ros/log/f129bbb8-dd8a-11e8-9d4c-5c514f3f2758/right_driver-2.log].
log file: /home/ytnpc2014b/.ros/log/f129bbb8-dd8a-11e8-9d4c-5c514f3f2758/right_driver-2*.log
-------------------------------------------------------------------------------

I already set the udev setting and tried to change the name from 10-kinova-arm.rules to 60-kinova-arm.rules.
In addition, I checked robot connection  with SDK.

What should I do?
I use Ubuntu 14.04, ROS indigo and connect through USB .

martine1406, 2018-11-01:
Hi

Have you changed the default [configuration file](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_bringup/launch/config/robot_parameters.yaml) for the robot? Is your connection type really configured to USB?

Are you using the code from the master branch?

kurenaloha, 2018-11-02:
Thank you for response.
I checked the default file and launched properly.

closed 2018-11-02 by kurenaloha