Kinovarobotics/kinova-ros #179: Unable to control kinova arm(j2s6s300) using rviz simulation

sakthiprasasd, 2018-11-08:
Hii friends,
      I am using ubuntu 14.04 and ROS -indigo to work on the Kinova arm(j2s6s300). The joint position control and Cartesian control,  I can do with command and using R viz simulation by selecting the topics in the update topic under the interactive marker control. But I can't able move the robotic arm according to the r viz simulation. After planning I gave the execute command arm start to move a little bit in the direction of destination, to reach the destination we need to click Execute  (in the Rviz simulation) several times. 
We need to press 'execute' or 'plan and execute' 10 times(not exactly) to reach the destination. 

The commands I used for that is given below

To launch Moveit and the real robot driver :
$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s6s300

and then
roslaunch j2s6s300_moveit_config j2s6s300_demo.launch

Is it because of any time-out issue? Can anyone please suggest a solution for this issue

martine1406, 2018-11-08:
I confirm that we cannot reproduce this issue at Kinova. If anyone ever experienced that behaviour, please let us know. To investigate if this is a problem with this specific robot unit, you can try to send angular velocity commands to the robot directly publishing on the j2s6s300_driver/in/joint_velocity, as suggested in [section ](https://github.com/Kinovarobotics/kinova-ros#velocity-control-for-joint-space-and-cartesian-space). Do you still have the same problem of the robot moving slowly?

sbrisebois, 2018-11-22:
I suspect timeout that occur between process on the PC. While running your trajectory, may rou open a console and run the 'top' command and see if the CPU is not used at 100%. This may lead to too much delay between commands to the Jaco arm and introduce a major slow-down.

sakthiprasasd, 2018-12-14:
We changed the entire Kinova software into a new system with better specification and we got the output. Thank you so much for your continuous support Martine

closed 2018-12-14 by sakthiprasasd

anubhav-dogra, 2021-01-17:
I am facing the same issue. while position action clients, actuators are running at good speed, but using velocity commands, the speed is very slow. 
Due to similar problem, mico arm is not able to reach is destined position using moveit. it just moves little bit and moves very slow.

anubhav-dogra, 2021-01-17:
Can you please help ?

anubhav-dogra, 2021-01-17:
> We changed the entire Kinova software into a new system with better specification and we got the output. Thank you so much for your continuous support Martine

How exactly you changed the entire software system ?