Kinovarobotics/kinova-ros #183: Old Jaco does not complete trajectory using MoveIt

MariotoA, 2018-11-19:
Hi, 
I'm working with an old JACO arm (v1) that falls under j2n6s300.
I'm using Ubuntu 16.04/ROS Kinetic after fixing packages kinova_moveit and more according to #134.
I bring up the robot and then I do:

`roslaunch j2n6s300_moveit_config j2n6s300_demo.launch`

Inside rviz I input this goal using default options:
[image]
The plan is generated successfully but in execution I get this output:
```
[ INFO] [1542645042.092347210]: Trajectory controller Receive trajectory with points number: 10
[ INFO] [1542645042.103220157]: Moved to point 0
[ INFO] [1542645042.445965850]: Moved to point 1
[ INFO] [1542645042.607131865]: Moved to point 2
[ INFO] [1542645042.815198707]: Moved to point 3
[ INFO] [1542645042.937967183]: Moved to point 4
[ INFO] [1542645043.109076274]: Moved to point 5
[ INFO] [1542645043.295111711]: Moved to point 6
[ INFO] [1542645043.453145399]: Moved to point 7
[ INFO] [1542645043.611233354]: Moved to point 8
[ INFO] [1542645043.994999491]: Moved to point 9
```
(on moveit terminal): (I enabled debug)
```
[ INFO] [1542645041.963804925]: Received new trajectory execution service request...
[New Thread 0x7fffa6abe700 (LWP 22293)]
[ INFO] [1542645042.065922625]: Joint_trajectory_action_server received goal!
[ INFO] [1542645042.066101559]: Joint_trajectory_action_server accepted goal!
[ INFO] [1542645042.066280982]: Joint_trajectory_action_server published goal via command publisher!
[ INFO] [1542645044.264533199]: Completed trajectory execution with status SUCCEEDED ...
[ INFO] [1542645044.264813333]: Execution completed: SUCCEEDED
[Thread 0x7fffa6abe700 (LWP 22293) exited]
[Thread 0x7fff727fc700 (LWP 22288) exited]
```
Everything seems to go well, but the arm stops midway.
I think this could be related to a time thing for each point, but I don't find exactly what .cpp files to look for or if it is in the drivers.
I would appreciate input on kinova_moveit working for older JACO versions and if urdf models are still valid.

Thanks for your time.

martine1406, 2018-11-19:
Hello @MariotoA 

Please first note that this package is not officially compatible with a Jaco V1, nor is it officially available for Kinetic. Now, for the work on kinetic, I generally pull the code and checkout to commit 94c25e1. Up to now, I had a successful build on Kinetic with that commit.

As far as Jaco V1 is concerned, there is no urdf specific to that robot. It is very similar to Jaco v2, so you can always use the urdf and moveit package for Jaco v2. If you want to adapt the urdf for Jaco V1 and create a moveit package, we may be able to give you some technical drawings to help you (please write to support@kinova.ca with that request). 

Still, I get the impression that the arm should not stop in the middle of the trajectory, even if it is a Jaco V1. Do you get the same problem when trying to move the robot using angular velocity commands (https://github.com/Kinovarobotics/kinova-ros#velocity-control-for-joint-space-and-cartesian-space)? Do you get the same problem no matter the trajectory you are trying to execute?

All the best,
Martine

martine1406, 2018-11-20:
Hi @MariotoA 

> I also notice the pid scripts are only for j2s7s300

What script are you talking about?

I think your issue might be related with this one: #172

Unfortunately, this issue seems to be related with using Kinetic (we do not have the same problem on Indigo). We have not investigated this issue actively, because we officially support Indigo. Nonetheless, it is interesting to know that you get a different behaviour with the commit I have suggested than with the fixes proposed in #134.

Let us know if you find the source of this issue. We'll do the same if we have time to do more tests on Kinetic with Moveit
All the best,
Martine

MariotoA, 2018-11-20:
I meant [these ones](https://github.com/Kinovarobotics/kinova-ros/tree/kinova-ros-beta/kinova_driver/src/scripts), they were on commit you referenced. I thought maybe they were required.

martine1406, 2018-11-20:
Oh ok yes. This is not official work from Kinova so it might only be available for a 7 dof robot. They are not required per se I think. They were developed to answer this issue: #116

6lue66, 2018-12-17:
I'm encounterng the same problem with a j2n6s300,I'm however using ros indigo on ubuntu 14.04. 

I also use : 

```
roslaunch j2n6s300_moveit_config j2n6s300_demo.launch
´´´

MariotoA, 2018-12-17:
Hi @6lue66,

I guess you are working on current master branch.
I've run a few tests on ubuntu 14 and indigo too and I've encountered the same problem.
I don't know your hardware specs but I had the opportunity to test on a better setting using U16/ROS Kinetic and results were similar to the ones on #169.

Good luck.

wxmerkt, 2020-10-29:
Did this issue get resolved? 

I am running into the same problem on a Jaco Gen v1 j2n6s300 equivalent. I manually installed the Kinova API v5.0.6 to be able to communicate over USB - and it works just fine for `joint_position_client` in `robot_control_modules` in `kinova_demo` and I can view the current state in Rviz. 

However, velocity commands (which is what the trajectory interface uses) are not being executed. The result of `result = kinova_api_.sendAdvanceTrajectory(kinova_joint);` in `KinovaComm::setJointVelocities` indicates `NO_ERROR_KINOVA`, yet, the arm does not move. I am using the example command to test: `rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 0.0, joint4: 10.0, joint5: 0.0, joint6: 0.0, joint7: 0.0}"`

Do you have any idea what the issue could be?

Many thanks in advance,
Wolfgang

felixmaisonneuve, 2021-06-02:
Hi Wolfgang,
I am not sure this issue applies to your case. Robots were moving and stopped unexpectedly half way. Your robot doesn seem to move at all.

Anyway, this issue has been inactive for over 2 years (exect from you of course), so I will close it.

If you still have issues with your arm, please create a new issue and I will be happy to help you

Thank you

closed 2021-06-02 by felixmaisonneuve