Kinovarobotics/kinova-ros #184: Reference frame with pose action client

GianlucaCerilli, 2018-11-19:
Hello,

I am trying to change the orientation (not the position) of the end-effector according to what I do with the following command:

**rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mdeg -- 0 0 0 0 20 0**

So, I can see the end-effector rotating around the Y axis (that apparently is not the one defined at pag.85 here https://www.kinovarobotics.com/sites/default/files/ULWS-RA-JAC-UG-INT-EN%20201804-1.0%20%28KINOVA%E2%84%A2%20Ultra%20lightweight%20robotic%20arm%20user%20guide%29_0.pdf).
The problem is that when I write this code, the end-effector rotates but in a different way. I think is something related with the reference frames, but I subscribe to the same topic, so I don't understand why I have different rotations. The code is:

>     end_effector_pose = rospy.wait_for_message('/j2n6s300_driver/out/tool_pose', geometry_msgs.msg.PoseStamped, timeout=3)
> 
>     end_effector_client = actionlib.SimpleActionClient('/j2n6s300_driver/pose_action/tool_pose', visual_servo_gianlu.msg.ArmPoseAction)
>     end_effector_client.wait_for_server()
> 
>     pose_to_reach = visual_servo_gianlu.msg.ArmPoseGoal()
>     pose_to_reach.pose.header = std_msgs.msg.Header(frame_id=('j2n6s300_link_base'))
>     pose_to_reach.pose.pose.position = geometry_msgs.msg.Point(
>         x = end_effector_pose.pose.position.x, y = end_effector_pose.pose.position.y, z = end_effector_pose.pose.position.z)
> 
>     end_effector_pose_radians = euler_from_quaternion([end_effector_pose.pose.orientation.x, end_effector_pose.pose.orientation.y, end_effector_pose.pose.orientation.z, end_effector_pose.pose.orientation.w])
> 
>     end_effector_pose_degrees = [rad_to_deg(end_effector_pose_radians[0]), rad_to_deg(end_effector_pose_radians[1]), rad_to_deg(end_effector_pose_radians[2])]
> 
>     increment_degrees = [0,20,0]    # Roll(X), Pitch(Y), Yaw(Z)
> 
>     next_pose_degrees = [end_effector_pose_degrees[0] + increment_degrees[0], end_effector_pose_degrees[1] + increment_degrees[1], end_effector_pose_degrees[2] + increment_degrees[2]]
> 
>     next_pose_radians = [deg_to_rad(next_pose_degrees[0]), deg_to_rad(next_pose_degrees[1]), deg_to_rad(next_pose_degrees[2])]
> 
>     next_pose_quaternion = quaternion_from_euler(next_pose_radians[0], next_pose_radians[1], next_pose_radians[2])
> 
>     pose_to_reach.pose.pose.orientation = geometry_msgs.msg.Quaternion(
>         x=next_pose_quaternion[0], y=next_pose_quaternion[1], z=next_pose_quaternion[2], w=next_pose_quaternion[3])
>
>     end_effector_client.send_goal(pose_to_reach)
>     end_effector_client.wait_for_result()

Thank you very much in advance!

martine1406, 2018-11-19:
Hi @GianlucaCerilli 

I think that rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mdeg -- 0 0 0 0 20 0 is not doing what you think it is doing. You would expect this command to rotate that gripper along what Y axis? What it is doing is changing the hand's orientation so the new orientation is equal to (the original Euler angle in X, the original Euler angle in Y + 20 degrees, the original Euler angle in Z). 

About the difference between your code and rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mdeg -- 0 0 0 0 20 0:
Can you try to monitor what is sent to the robot in each case. The function [setCartesianPosition](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_comm.cpp#L1036) is called by [kinova_tool_pose_action.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_tool_pose_action.cpp)

GianlucaCerilli, 2018-11-22:
Hi @martine1406,

Thank you very much for your reply. Yes, I misunderstood what the pose_action_client does. So, I expect  from the end-effector to change the relative pose translating/rotating wrt base reference frame.
In the end I have managed to correct the error in the script, thank you.

At the moment I am fighting with a problem derived from the mismatch between the real robot pose and the one that I see in Rviz (e.g. when I run j2n6s300_demo.launch). I am extrinsically calibrating a camera mounted on the end_effector and this mismatch leads the process to a wrong result.

You already told me in a previous conversation that you officially support indigo (I am on kinetic), so I don't want to bother you, but maybe in this case you can easily understand what is the problem and it would be really helpful for me to go ahead with the code, since I have a very important deadline.

I have updated the urdf as you told me one month ago, so now I have 
>     <xacro:property name="joint_end_effector_origin_rpy" value="${J_PI} 0 ${J_PI/2}" />
but I have also tried with:
>     <xacro:property name="joint_end_effector_origin_rpy" value="${J_PI} 0 0" />
but this changes only the reference frame. It does not change the pose of the gripper in the simulator, that is different from the real one. I am also using movegroup: arm, but even if I use movegroup: gripper nothing changes...but maybe this is not related with my problem.

Thank you very much in advance

martine1406, 2018-11-22:
Hi @GianlucaCerilli 

The home position on your real robot seems off. The problem seems to be on your 6th joint, which does not have the right zero. Have you done a reset zero position on it? If so, you positioned the robot in the wrong position on joint 6.

What you'll need to do is to redo a zero position as explained in the [user guide](https://www.kinovarobotics.com/sites/default/files/ULWS-RA-JAC-UG-INT-EN%20201804-1.0%20%28KINOVA%E2%84%A2%20Ultra%20lightweight%20robotic%20arm%20user%20guide%29_0.pdf), p. 70. When you send a 180 degree command on all joints, you should see your robot looking like the second picture on that page.

It should solve your issue. Let me know
Martine

GianlucaCerilli, 2018-11-22:
Hi @martine1406 ,

Thank you very much. When I run **rosrun kinova_demo joints_action_client.py -v j2n6s300 degree -- 180 180 180 180 180 180** the robot assumes this pose, that is incorrect:

[image]

But how can I do now to first, rotate the end_effector precisely until the correct pose, and then, to save the new zero pose? Because in the user guide I only see what is the correct zero pose.

Thank you,
Gianluca

martine1406, 2018-11-22:
yes that's it. You try to align the end effector as precisely as you can to reproduce the configuration shown in the user guide, then indicate to the robot to reset its zero position on the sixth joint (address is 21, then click Apply)

[image]

GianlucaCerilli, 2018-12-05:
Hi @martine1406,

As you already told me the rotation of the end-effector with pose_action_client rotates it not wrt to a particular frame, but just to increase/decrease the angle of the desired degrees wrt to the base frame.
What I would like to do instead, is to actually rotate (no translation) the end-effector around its own x and/or y axis (because I have a camera mounted on it and I would like to pan/tilt it according to what I see). Is it possible somehow?

I also wanna thank for your replies in these months. You have been always very kind and helpful and since I am almost done with my work I think I will not bother you much more again!:)

martine1406, 2018-12-06:
Hi @GianlucaCerilli 

When using pose_action_client, you need to specify the position of the end-effector (as an XYZ vectorO and the orientation of the end effector (as XYZ Euler angles) with respect to the base frame. This is the only thing the robot can understand. If you want to control the rotation of the end effector in its own frame (camera frame), you need to make yourself a small conversion algorithm that will transform your user command given in the camera frame to the robot's base frame (it comes down to multiplying your command by a rotation matrix. You can get this rotation matrix by monitoring the end effector's orientation, which is given with respect to the base frame). You have no mode to provide your pose command in the camera frame and have the robot automatically interpret it unfortunately. But this [code](https://drive.google.com/a/kinova.ca/file/d/1B1jknPxJT03AT69l6HY0uj-xJh995Jyw/view?usp=sharing) (using the Kinova API directly and the Eigen library, not the ROS package) can give you a rough idea on how to do it.  The code uses velocity commands, but it is the same idea for position commands. 

Thank you very much for your good words!
Martine

closed 2019-01-16 by nsaif-kinova