Kinovarobotics/kinova-ros #189: Finger Positions using Low-Level API while doing Torque Control

vinaysriram, 2018-12-03:
I am using the Kinova Low-Level API, and wish to set the finger positions while in torque control mode. The only way I could find to set the finger position was in KinovaComm::setFingerPositions, which shows how to do it when using either angular position control type or Cartesian position control type. Is there a way to set finger positions while using torque control? Thanks in advance for your help.

martine1406, 2018-12-03:
Hi @vinaysriram 

You say you are using the low-level API. This means you are directly sending RS485 messages to the actuators. Can you first confirm that information?

Then, please consider that the fingers can only be controlled in position or velocity mode (the finger actuators do not respond to torque commands) when using the high-level API. Using the low-level API, the finger actuators can only be controlled in position. While sending torque commands to the arm's rotary actuators, it is possible to send position commands to the fingers. 

Please give me more details on your implementation and I'll try to guide you at best,
Martine

vinaysriram, 2018-12-14:
Hi,

Sorry for the delayed follow-up. Is there some example code supporting this? I am looking at lines 1328 to lines 1412 of this file: [https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_comm.cpp]. Here, the allowed control modes are cartesian and angular, and the positions of the fingers are set through the `sendBasicTrajectory`API call. Is there a way to use the functions defined in [https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp] to set the finger positions while controlling Torques?

Thank you for your help.

- Vinay

martine1406, 2018-12-15:
Hello @vinaysriram 

Thank for your follow-up. First of all, you did not answer my question. 
You say you are using the low-level API. This means you are directly sending RS485 messages to the actuators. Can you first confirm that information? If that's the case, I invite you to communicate with us at the address: support@kinova.ca. This is a place to report issues with the ROS package, not the API, and especially not the low-level API
In order words, this ROS package does not support the low-level API. Can you confirm what you are using?

Now, if you want to use the high-level API instead, the only way to control the fingers right now while the robot is in torque mode is to use the joystick. To do this programmatically, you need to emulate the joystick using a virtual joystick and virtual joystick commands. I am on maternity leave, so I'll let my colleagues explain you more how to do this. But basically, here is a code snippet that shows how to use virtual joystick commands with the Kinova C++ API (working on my side). 

Here is a snippet of example code that works on my side. Don't forget to also Init the fingers :-) 
			`//switch to torque mode
			result=MyInitFingers();
			MySwitchTorque(TORQUE);

			cout << "Here you have 10 seconds to play with the robot in torque mode." << endl
				<< "Try to move it with your hand" << endl << endl;
			Sleep(10000);

			ControlMappingCharts ans;
			MyGetControlMapping(ans);

			result = (*MyStartControlAPI)();

			JoystickCommand virtualCommand;
			//Initializing the command.
			for (int i = 0; i < JOYSTICK_BUTTON_COUNT; i++)
			{
				virtualCommand.ButtonValue[i] = 0;
			}
			virtualCommand.InclineForwardBackward = 0;
			virtualCommand.InclineLeftRight = 0;
			virtualCommand.MoveForwardBackward = 0;
			virtualCommand.MoveLeftRight = 0;
			virtualCommand.PushPull = 0;
			virtualCommand.Rotate = 0;
			virtualCommand.Rotate = 0;
			AngularPosition ang;
			cout << "Sending trajectory" << endl;
			//do only once after a reset
			virtualCommand.ButtonValue[1] = 1; //to switch to finger mode
			for (int pp = 0; pp < 10; pp++)
			{
				(*MySendJoystickCommand)(virtualCommand);
				Sleep(5);
			}
			virtualCommand.ButtonValue[1] = 0;

			//This is the point that will be send to the robot.
			//We prepare the virtual joystick command that will be sent to the robotic arm.
			cout << "We take control of the robotic arm with an opening/closing motion of the hand." << endl;
			for (int pp = 0; pp < 200; pp++)
			{
				virtualCommand.InclineLeftRight = 1;
				(*MySendJoystickCommand)(virtualCommand);
				Sleep(5);
			}
			for (int pp = 0; pp < 200; pp++)
			{
				virtualCommand.InclineLeftRight = -1;
				(*MySendJoystickCommand)(virtualCommand);
				Sleep(5);
			}
			Sleep(8000);

			cout << "Back to Trajectory mode" << endl << endl;
			MySwitchTorque(POSITION);` 

[https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp] most of the functions from the Kinova API. You may need to add the missing ones and write the appropriate code in https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_comm.cpp and https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp to call your new functions from ROS. 

I hope this helps

vinaysriram, 2019-02-26:
Following up on the above issue - yes, we are actually using the high-level API (the one given in [https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp]) . Using this API, the provided code does not seem to be working - specifically, the arm does not seem to respond to joystick commands sent from the emulated joystick. 

We are using a Kinova Mico with 2-finger gripper. Is there anything specific for initialization that is missing from the above code?

Furthermore, is there documentation that describes how set the fields of a JoystickCommand for particular actions? Thanks very much for your time.

nsaif-kinova, 2019-02-26:
Hello Vinay,
Please let me help you with this issue.
Are you able to move the fingers using the joystick?
Did you initialize the fingers? to do that you have to send the robot home and open the fingers completely so you can use them before you switch to torque.

You can also do the following test :
1-Turn on the robot
2-Send robot home
3-Open fingers completely
4-Using Torque Console switch to torque mode
5-Using joystick you should be able to open and close the fingers.
Please let me know if this will help you fix the problem.
Sincerely,

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over a years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve