Kinovarobotics/kinova-ros #191: process has died after calling void moveit::planning_interface::MoveGroupInterface::stop()

ChanCheryl, 2018-12-07:
hi, I want to stop the robot arm immediately during execution. But when I called the moveIt API like "gripper_group_.stop(); group_.stop()", the kinova driver node died. What's the problem? Any advice would be appreciated. 

[image]

samuelhovington, 2018-12-07:
Hi ChanCheryl,
Can you, please, provide the exact command line that you tried for both moving the arm and stop it?
Also, can you confirm that you use ros indigo?
From the picture it looks like you are controlling a real arm. If so, tried to stop it with the service /'${kinova_robotType}_driver'/in/stop instead of moveit function.

For your arm it should be : 
`rosservice call /j2n6s300_driver'/in/stop`

Sincerely,

thekillersep, 2018-12-08:
Hi @samuelhovington 
Thank you very much for your reply. My ROS version is indigo. For quick test, I made some changes in your official demo code pick_place.cpp .I just added a subscriber ,a message callback function and  published the corresponding message in terminal.The exact command line and added code are shown as follows.
After I encountered this problem several days ago,I did the same thing like what you said using the /j2n6s300_driver/in/stop  service instead of moveit function. It works but the robot arm will stop and wait until I call the /j2n6s300_driver/in/start service.
However I want to stop the robot arm immediately and accept new goals right away.I do not want to wait.Just stop and move to the next new pose.Do you provide a  corresponding service in this package?
Here is the description of "void moveit::planning_interface::MoveGroupInterface::stop()" :stop any trajectory execution, if one is active.
I did the same thing without connecting the real arm in Rviz. It worked.Once I controlled the real arm, process has died.
[image]

thekillersep, 2018-12-11:
Hi,@samuelhovington
The robot will stop instantly when we call the stop service.Then it's ok to call the start service and send a new pose to the arm.I have already tested it last week.
But that's not what I meant.I write a simple action server and client.The client will calculate real-time pose of the object.The server will receive the goal and move to the object.If object pose changes,client will send the new object pose .If a new goal comes when the robot arm is moving to the current goal,the action server will call "void preemptCB()" to stop the robot arm and set the action server state to preempted.Then go back to "void goalCB()" to finish the current goal.Although I determine the action server state in "void goalCB()",it won't help.The robot arm will stop and wait until the current goal execution time is over.Then the action server will call "void goalCB()" to process the new coming goal.Accorrding to what you said,I should call the stop service to stop the arm in "void preemptCB()",then call the start service to start it in "void goalCB()". That's  what I did too.The code is shown as follows.
[image]
However the robot arm will wait maybe it's not too long.You can see it from the picture below.Is it possible to kill the waiting time between "Arm stop requested" and "Arm start requested"?What do you think?Thank you for your help.
[image]

samuelhovington, 2018-12-11:
Unfortunately, Kinova API does not have a method to do the exact thing you want to do, because it is 
Moveit! that send the point to the arm. The method from Moveit! API should work, but I can't figure it out why it does not. 
I look at it during the next days and I will let you know If I can call the void moveit::planning_interface::MoveGroup::stop (void) successfully. 

Let me know if I can do something else.

martine1406, 2018-12-11:
Hi all

What if you try to send a new active goal without cancelling the previous one? Won't the new one automatically replace the old one?

thekillersep, 2018-12-14:
Hi, @martine1406 
I am afraid this is not going to work.I tested like what you said and failed.Using the method from Moveit API,we need to stop the robot arm firstly,then send a new active goal.The new active goal will do nothing during execution without cancelling the previous one.The robot arm will still execute the previous goal.
@samuelhovington Do you encounter the same problem and find a way to avoid it?Waiting for your good news!Thanks all of you.

thekillersep, 2018-12-18:
Hi , @martine1406 
It doesn't work.Still wait between "Arm stop requested" and "Arm start requested". The moveit function doesn't work with the old problem.Thanks for your help.And I'll try to find another way to avoid the problem.

martine1406, 2018-12-18:
Hi

Hmm okay I see. I think it is because the way [kinova_joint_trajectory_controller](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_joint_trajectory_controller.cpp) is defined. [joint_trajectory_action_server.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp) might be able to cancel the active goal and replace it by the new one it just received, but when it publishes on "/"+ robot_name + "_driver/trajectory_controller/command", the listener in [kinova_joint_trajectory_controller](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_joint_trajectory_controller.cpp) is still busy treating the active goal. 

In all cases, let us know if you find an elegant way to implement what you need.
Good luck

closed 2019-01-15 by nsaif-kinova

iretiayo, 2019-03-04:
@thekillersep, did you find a solution to the problem? I am having the same issue here and thought to check if you found a solution. Thanks.

thekillersep, 2019-03-05:
Hi, @iretiayo  @samuelhovington  @martine1406
Sorry for that I didn't share my solution timely.
After calling the void moveit::planning_interface::MoveGroupInterface::stop() function, the trajectory controller will receive trajectory with points number:0. Then process has died. So check the size of trajectory command points in void JointTrajectoryController::commandCB() function in kinova_joint_trajectory_controller.cpp firstly.Then a member flag "stop_" has been added and set to false in constructor,which is used to clear the rest of trajectory command points and stop the timer to publish joint velocity in void JointTrajectoryController::pub_joint_vel() function. If the size of trajectory command points is 0,"stop_" flag will set to true and the void JointTrajectoryController::commandCB() function just return. In void JointTrajectoryController::pub_joint_vel() function, check the "stop_" flag to decide whether to stop the timer or not.The code in the red rectangle is what I added and you can see the result from the pictures below. I also implemented it by /j2n6s300_driver/in/stop service instead of moveit function, which is a little more complicated but with the same method as I said above.Let me know if it works in your experiment .Good luck.
[image]
[image]
[image]

iretiayo, 2019-03-08:
Thanks @thekillersep for your prompt and detailed response. I tried out your fix and it worked i.e. it now handles `stop trajectory` nicely.
You mentioned that you also implemented it similarly using _/j2n6s300_driver/in/stop service_; does that have any speed/performance benefit?
Thanks again.

thekillersep, 2019-03-10:
Hi, @iretiayo  @samuelhovington @martine1406 
The  /j2n6s300_driver/in/stop service is for an emergency stop. The robot arm does not accept new goals after having been stopped until call the j2n6s300_driver/in/start service to start the arm.While the moveIt stop() function is used to stop any trajectory execution, if one is active. 
The robot will stop instantly when call the stop service. However the robot arm will still wait between "Arm stop requested" and "Arm start requested". You can see it from the picture below.
[image]
In order to stop the arm and execute the new active goal immediately, we should make the same changes like what we did using moveIt stop() function. During calling the stop service, the member flag "stop_"  will set to true to clear the rest of trajectory command points and stop the timer to publish joint velocity. During calling the start service, the member flag "stop_"  will set to default false. Besides, we need to mark the current goal as aborted "active_goal_.setAborted()", set the flag has_active_goal_ to false and make the void JointTrajectoryActionController::goalCBFollow(FJTAS::GoalHandle gh) function just return in https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp. You can see the result using stop service from the picture below .
[image]
I think there is no speed/performance benefit using the stop service. Anyway we will get the same result and the robot will handles stop trajectory nicely. The easiest solution I see is call the moveIt stop() function to stop the arm. What do you think?Let me know if I can do something else.