Kinovarobotics/kinova-ros #192: Where is hand axis?

kurenaloha, 2018-12-08:


samuelhovington, 2018-12-10:
Hi kurenaloha,

Actually, there is two way to show hand orientation:
Quaternion convention with rostopic echo /j2n6s300-driver/out/tool_pose
Cartesian convention with rostopic echo /j2n6s300_driver/out/cartesian_command 

The left panel should show the exact same thing as the /j2n6s300-driver/out/tool_pose topic like it is on the following picture
[image]

Yes, both the left panel values and the terminal one use the world frame.

Let me know if you need extra information.

Sincerely,

kurenaloha, 2018-12-11:
Thank you very much!!! 
I understood 2 ways to show hand orientation！

However, I can't understand the hand axis... 
Whenever I send "Theta z" to jaco, only the 6th joinｔ moves. 
ex.  rosrun kinova_demo pose_action_client.py -v -r  j2n6s300 mrad -- 0 0 0 0 0 1
I think if the hand uses the world frame,  it is rotated around the z-axis of world frame...

I want to rotate the hand around  the z-axis of world frame. How should I do?

samuelhovington, 2018-12-11:
The hand uses the world frame to translate in space, but it uses its own frame to rotate in space. The z-axis will always point forward from the wrist view. X and Y axis will always be in the same direction if you look reference to the fingers as shown below
[image]

This said, It does not have a function that does exactly what you want to do, but can you explain why you want to control the arm around the z-axis of the world frame? 

Maybe I can suggest you something else or help you doing it.

martine1406, 2018-12-11:
like @kurenaloha, I was under the impression that 

rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mrad -- 0 0 0 0 0 1

would rotate the hand around the base's Z axis. This sends a Cartesian position command to the robot, not a Cartesian velocity command. For Cartesian velocity commands, I agree that the Jaco uses the base frame for translation and hand frame for rotation. But not for position commands. For position commands, the position is expressed as a vector with respect to the base axis and the orientation is expressed as XYZ Euler angles, again with respect to the base frame. 

Unfortunately, I do not have a real Kinova arm to test with, but after looking at the code, thius behaviour still seems weird to me. Can someone test without the relative (-r) option. i.e. get the current robot pose, modify it slightly and send the new pose explicitly?

rosrun kinova_demo pose_action_client.py -v  j2n6s300 mrad -- (new pose close enough from the current pose not expressed as a relative motion)

kurenaloha, 2018-12-12:
@samuelhovington 
Thank you for your reply.

Actually, I want to know and record how much the hand is moved from initial position and orientation, then I want to move it as much as recorded from different initial position and orientation. 
To do so, I think I should use the base frame...

What do you think about it?

samuelhovington, 2018-12-12:
@kurenaloha 

Ok, if I understand well what you are trying to do, you want to record a trajectory referenced to a specific point in space and then reproduce the exact same trajectory, but from another point in space. Is that right?

If it is, I see a quite easy way to do it. 

First of all, the position in XYZ is always referenced to the world frame so you just have to add an offset to all points in the trajectory you have recorded. This offset represents the difference between your second point of reference and the first one. By example, Your second point of reference could be where you want the second trajectory to start and the first one could be the first point of your previous recorded trajectory.

For the orientation, it is the same procedure except you use XYZ orientation instead of XYZ position. The fact that the orientation is referenced to the end effector itself does not change anything.
Note that you have to send your trajectory without the -r option. you also need to consider that each point on the new trajectory must be reachable and not too close to a singularity 

I tried to be as clear as I can, but let me know if you need extra information.

kurenaloha, 2018-12-13:
@martine1406 
Thank you for your reply.

I tested without the relative (-r) option. The following video shows the result.
[drive-download-20181213T014504Z-001.zip](https://github.com/Kinovarobotics/kinova-ros/files/2674224/drive-download-20181213T014504Z-001.zip)

I think XY orientaiton use the world frame and only Z orientation uses hand frame.

Actually, I want to know and record how much the hand is moved from initial position and orientation , then I want to move it as much as recorded from different initial position and orientation.
In the other words, I want to record trajectory and send it to jaco whose initial position and orientation are different.

What should I do?

kurenaloha, 2018-12-13:
@samuelhovington 
Thank you for your reply.

Yes, I'm trying to do that.
I understood the way for the position in XYZ because of using world frame.
However, I think  XY orientaiton use the world frame and only Z orientation uses hand frame...

[drive-download-20181213T014504Z-001.zip](https://github.com/Kinovarobotics/kinova-ros/files/2674264/drive-download-20181213T014504Z-001.zip)

SimonMichaudKinova, 2018-12-19:
Hello @kurenaloha,
 
We looked at your problem and we identified that the Euler angles representation of the orientation of the end effector is given with respect to the base frame in cartesian position mode, and with respect to the end effector frame for velocity position.

As for the XY and Z orientation difference, this behavior is due to the Euler angle convention that we use. For intrinsic rotations, X rotations are aligned with the X axis of the base, the Y rotation is aligned with an intermediate frame (often simillar to the y-axis of the base) and the Z rotation is aligned with an other intermediate frame that often aligns its Z-axis with the Z axis of the end effector. This is why you observe this behavior.

For your application of moving a trajectory to different points in space, we suggest you to create trajectories in cartesian velocity, because most movements can be translated in space. We do not recommand using cartesian positions because it is more restrictive for larger translations. The robot will often not be able to reach points in its trajectory in cartesian position if this trajectory is translate in XYZ. In every cases, a translation that is too large can always be problematic.

We hope that this will help.

closed 2019-01-15 by nsaif-kinova