Kinovarobotics/kinova-ros #193: Tool coordinate control of Kinova J2s6s300

sakthiprasasd, 2018-12-14:
Hi,
             Anyone can suggest a method or tutorial for tool coordinate control of Kinova arm J2S6S300??

sakthiprasasd, 2018-12-14:
Hi Nassif,
               Tool coordinate control is different from cartesian. We did the Cartesian control, but no provision is there to control the arm in tool coordinate system. Please refer to the image to get a better idea
[image]

sbrisebois, 2018-12-14:
Hello  Sakthiprasasd,

My name is Sylvain Briseboi, a colleague of Martine and Nassif.

I think I understand what you want and here is an implementation that
Martine gave me to perform a tool motion control on the Jaco.

// Do the magic to convert the joystick command into a Gripper Ref mode
commande
                        CartesianPosition pose;

                        // Get actual gripper pose (Position & orientation)
                        lApiErr = m_RobotAPI.fGetCartesianCommand(pose);
                        KeDelay::SleepMs(5);

                        if(lApiErr != NO_ERROR_KINOVA)
                        {
                            LOG(LOG_ERR, "Robot API SendJoystickCommand
Error: %d",lApiErr);
                            m_ConsecutiveCommErrors++;
                        }
                        else
                        {
                            //the transformation matrix between
end-effector and base frame is given by the end-effector orientation
                            Eigen::Matrix3f orientation_robot;
                            orientation_robot =
Eigen::AngleAxisf(pose.Coordinates.ThetaX, Eigen::Vector3f::UnitX())
                                                *
Eigen::AngleAxisf(pose.Coordinates.ThetaY, Eigen::Vector3f::UnitY())
                                                *
Eigen::AngleAxisf(pose.Coordinates.ThetaZ, Eigen::Vector3f::UnitZ());

                            //translation vector in end-effector frame
                            Eigen::Vector3f
translation_vector_ee_frame(lthreadJoystickCommand.InclineLeftRight,

-lthreadJoystickCommand.InclineForwardBackward,

lthreadJoystickCommand.Rotate);

                            //translation vector in base frame
                            Eigen::Vector3f translation_vector_base_frame =
orientation_robot * translation_vector_ee_frame;

                            //update the x-y-z part of the command
                            lthreadJoystickCommand.InclineLeftRight =
translation_vector_base_frame(0); // X
                            lthreadJoystickCommand.InclineForwardBackward =
translation_vector_base_frame(1); // Y
                            lthreadJoystickCommand.Rotate =
translation_vector_base_frame(2); // Z
                        }

This algo may need some modification, the idea is there. It will allow to
control the translation of the end effector in the gripper frame.
The orientation still need to be adapted if you want it to be controlled in
position, but in velocity it's already in the gripper frame.

Let me know if you need assistance for that too.

Have a nice day,
Sylvain

Sylvain Brisebois Ing.

Expert d'application client - division Innovation

Field application expert - Innovation division

514-277-3777 ,4078

514-244-8238

sbrisebois@kinova.ca

4333 boul. de la Grande-Allée, Boisbriand (QC)
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

<https://maps.google.com/?q=4333+boul.+de+la+Grande-All%C3%A9e,+Boisbriand+(QC)++%C2%A0+J7H+1M7&entry=gmail&source=g>J7H
1M7
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

kinovarobotics.com <http://www.kinovarobotics.com/>   Facebook
<http://facebook.com/kinovarobotics/>  Twitter
<http://twitter.com/kinovarobotics>  LinkedIn
<https://www.linkedin.com/company/kinova/>  YouTube
<https://www.youtube.com/user/Kinovarobotics>

The information contained in this email message is private and
confidential. It is intended for the exclusive use by the recipient named
above. If you are not the intended recipient, be advised that disclosing,
copying, distributing or taking any action in reliance on the contents of
this information is strictly prohibited. If you have been sent this email
by error, please destroy it and immediately inform us at
info@kinovarobotics.com.

sbrisebois, 2018-12-14:
Hello  Sakthiprasasd,

My name is Sylvain Brisebois, a colleague of Martine and Nassif.

I think I understand what you want and here is an implementation that Martine gave me to perform a tool motion control on the Jaco.

This example is intended to apply a transformation to a joystick command to control the translation in the gripper frame instead of the base frame.

// Gripper frame control
                        CartesianPosition pose;

                        // Get actual gripper pose (Position & orientation)
                        lApiErr = m_RobotAPI.fGetCartesianCommand(pose);
                        KeDelay::SleepMs(5);

                        if(lApiErr != NO_ERROR_KINOVA)
                        {
                            LOG(LOG_ERR, "Robot API SendJoystickCommand Error: %d",lApiErr);
                            m_ConsecutiveCommErrors++;
                        }
                        else
                        {
                            //the transformation matrix between end-effector and base frame is given by the end-effector orientation
                            Eigen::Matrix3f orientation_robot;
                            orientation_robot = Eigen::AngleAxisf(pose.Coordinates.ThetaX, Eigen::Vector3f::UnitX())
                                                * Eigen::AngleAxisf(pose.Coordinates.ThetaY, Eigen::Vector3f::UnitY())
                                                * Eigen::AngleAxisf(pose.Coordinates.ThetaZ, Eigen::Vector3f::UnitZ());

                            //translation vector in end-effector frame
                            Eigen::Vector3f translation_vector_ee_frame(lthreadJoystickCommand.InclineLeftRight,
                                                                        -lthreadJoystickCommand.InclineForwardBackward,
                                                                        lthreadJoystickCommand.Rotate);

                            //translation vector in base frame
                            Eigen::Vector3f translation_vector_base_frame = orientation_robot * translation_vector_ee_frame;

                            //update the x-y-z part of the command
                            lthreadJoystickCommand.InclineLeftRight = translation_vector_base_frame(0); // X
                            lthreadJoystickCommand.InclineForwardBackward = translation_vector_base_frame(1); // Y
                            lthreadJoystickCommand.Rotate = translation_vector_base_frame(2); // Z
                        }

This algo will need some modification for your situation, but the idea is there. 
The orientation still need to be adapted if you want it to be controlled in position, but in velocity it's already in the gripper frame.

Let me know if you need assistance for that too.

sakthiprasasd, 2018-12-15:
Hi Sylvain Brisebois,
                             Thank you so much for your reply. We need some more support from your side. The first time we are using visual studio and Kinova codes. We don't know where we need to modify or which file we need to change to get tool coordinate control.
 In windows, Example_cartesian mode control  - 3 header files and one source code are there in that the c++ source code eigon vector, frame details, and DH parameters are not there. We cant able to find the details of functions mentioned in that program. From where we get those details?
                                 Through SDK  we can able to run examples but not able to run through the visual studio. Is there any additional thing we need to do to get the arm movement??

sakthiprasasd, 2018-12-17:
Hi Sylvain Brisebois,
                                 If we modify the urdf,  we can find the solution for this issue? using SolidWorks it's very difficult to do the modification of urdf.  what software did you use to do the urdf?

 Kinovarobotics/matlab_Kinovaapi_wrapper is there to integrating Kinova with Matlab. Through Matlab, we can achieve tool coordinate control ?? Is there any example in Matlab which is helpful for us?

samuelhovington, 2018-12-17:
Hi sakthiprasasd, 

Yess the Kinovarobotics/matlab_Kinovaapi_wrapper is there to integrating Kinova with Matlab. 
The following codes will help you to achieve your goal. 

[tool_pose.txt](https://github.com/Kinovarobotics/kinova-ros/files/2687659/tool_pose.txt)
[EulerXYZtoRot.txt](https://github.com/Kinovarobotics/kinova-ros/files/2687660/EulerXYZtoRot.txt)

This code is ready to use with the matlab wrapper from kinova just rename them as matlab files. 
This code, make the arm move in hand effector frame, but only in transaltion. Fortunately, when you control  the arm in cartesian velocity for rotation, it already move in end effector frame. This means that you only can use the sendCartesianVelocityCommand as is for End effector rotation.

Sincerely,

sakthiprasasd, 2018-12-18:
Hi Sylvain Brisebois,
                         Thank you so much for your support. Currently, we are working in Matlab. We can able to control the arm. Tool coordinate control is not completely working we are trying to solve the issues. 

Is there any method to disable the singularity avoidance method?

samuelhovington, 2018-12-18:
To disable the singularity avoidance, you have to control the arm with the low-level API.
This API is not supported in Matlab. 

For tool control, which issues are you facing? Is it about the code I provide you?

sakthiprasasd, 2018-12-20:
Hi Sylvain Brisebois,
Thank you so much for your support. Tool coordinate is working with Matlab. But we are doing everything in ROS platform, so can you give the ROS package for tool coordinate control?

samuelhovington, 2018-12-20:
Hi sakthiprasasd,

We do not have a package ROS for tool coordinate control, but the logic behind the Matlab code is the same that the one you have to implement in ROS. 
You have to create your own node which passes the translation command through the transformation matrix, given in Matlab code, before sending it to the arm. For rotation, you only have to send velocity command. 
To summarize, your node will be an intermediate between the user and the Kinova ROS package. It will take tool coordinate command in input from the user and will output base coordinate command that will be sent directly to the arm via the API.

Let me know if I can do something else.

samuelhovington, 2018-12-20:
One Thing I can suggest you is to modify the pose_action_client.py. This node already calls each function of the API that you'll need to control the arm in tool coordinate.

sakthiprasasd, 2018-12-21:
Hi Sylvain Brisebois,
Thank you so much for your support. Now we can able to control the arm in 'tool coordinate system' with the help of the commands.

closed 2018-12-21 by sakthiprasasd