Kinovarobotics/kinova-ros #195: Velocity control of kinova arm j2s6s300 using joystick remotely

sakthiprasasd, 2018-12-22:
Hi,
      We want to control the Kinova arm (J2s6s300) remotely using a joystick in ubuntu. For that, we used the package 'joy2key'. Using this method we can not able to control the velocity, which is possible in Kinova joystick. Through the joystick, which we are connecting to the laptop can only map buttons to the alphabets assigned in the virtual joystick,  so we cannot get the speed control. Using any other package or method, can we able to solve this issue?

sbrisebois, 2019-01-03:
Hi

sbrisebois, 2019-01-03:
Hi Sakthiprasasd,

I recommend you to have a look at the movo teleop that allow to control the robot in speed : 
https://github.com/Kinovarobotics/kinova-movo/blob/master/movo_common/movo_ros/src/movo/movo_teleop.py

It's very simple to use and will allow you to create customs joystick mapping as you want.

closed 2019-01-16 by nsaif-kinova