Kinovarobotics/kinova-ros #198: SSH port is not running in Kinova j2s6s300 robotic arm

sakthiprasasd, 2019-01-01:
Hi,
    We tried to control the Kinova arm (j2s6s300)  using an ethernet cable. It is working fine, but we cannot access the Kinova controller using putty. One of the network experts in our institution said that the SSH port is not working as a reason for this.  What is the actual reason for this issue? If it is related to SSH port, what is the solution for that?

sbrisebois, 2019-01-03:
Hi,
The Kinova arm (j2s6s300) doesn't have an SSH server running in it's controller. This is why attempting to open an ssh session fail. The only way to communicate with the arm is to use the API provided by the SDK.

Sincerely,
Sylvain

sakthiprasasd, 2019-01-04:
Hi, 
       Thanks for your reply. We saw in one your user guides(attached to this message) the wireless communication using the wifi adapter. In that, they are accessing the Kinova controller. Please tell me how they are accessing, they are using putty only. The same wifi adapter is required of the wifi accessing? The wifi adapter mentioned in the user guide, can we purchase that separately? 

[KINOVA_Remote_robotic_manipulation_system_User_Guide.pdf](https://github.com/Kinovarobotics/kinova-ros/files/2726135/KINOVA_Remote_robotic_manipulation_system_User_Guide.pdf)

sbrisebois, 2019-01-04:
Hi,

You are referring to the "Remote robotic manipulation system" and it's not using wifi, but a propritary wireless solution and the SSH connection is done on the adapter box, not on the arm controller. This kit is intend to control a Jaco2 with a XBOX gamepad on a wireless link. It also enable to control the Jaco2 from a tool based reference.

sakthiprasasd, 2019-01-04:
Hi,
     Thanks for your reply. Is it possible to purchase the adapter box separately? If it is possible whom should I need to contact for the same?

sbrisebois, 2019-01-04:
Hi,

Yes you can buy one adapter box. I just want to precise that the adapter box is intended for manipulating the ARM via the XBOX gamepad, no development can be done inside for the moment. In the future this product may be used for developing, but currently this is not supported. So if you want to run your code on it, you should check for a Raspberry Pi or something similar. If you only need to control the arm, then yes it's a good product ! You can contact me (sbrisebois@kinova.ca) and I will direct you with our sales dept.

closed 2019-01-16 by nsaif-kinova