Kinovarobotics/kinova-ros #202: Moveit for m1n4s200

2016mez0019, 2019-01-03:
How to make moveit config for m1n4s200 ? 
Moviet setup assistant is not accepting m1n4s200.xacro from kinova_description/urdf.
I have actual mico 4 dof 2 fingered arm, and i want to run it using moveit-rviz.

nsaif-kinova, 2019-01-04:
Hello Anubhav,
Did you follow the instructions on this link? https://github.com/Kinovarobotics/kinova-ros/wiki/MoveIt
Can you please post the commands you are using?
Sincerely,

anubhav-dogra, 2019-01-04:
Hello Nsaif

Config files for m1n4s200 is not available in "kinova-ros/kinova_moveit/robot_configs/"
So to create one -
cd catkin_ws/
source devel/setup.bash

roslaunch moveit_setup_assistant setup_assistant.launch

 Then - create new moveit config package.

choosing file from catkin_ws/src/kinova-ros/kinova-description/urdf/m1n4s200.xacro

[image]
then
[image]

What should I do?

alexvannobel, 2019-01-17:
Hi @anubhav-dogra ,

Really sorry for the long delay.

I just tested this and the MoveIt! Setup Assistant won't take let you change command line arguments for xacro files in the Indigo version.

You will have to translate the m1n4s200_standalone.xacro file to a URDF file using the xacro node using the following command (you may want to rename the default robot_root xacro property to something else to match your own root's name in the xacro file beforehand):
```
cd path/to/your/catkin_ws/src/kinova_description/urdf
rosrun xacro xacro.py m1n4s200_standalone.xacro > m1n4s200.urdf
```
This static URDF file should then properly be loaded into the MoveIt! Setup Assistant.

Hope this helps,

Best,
Alex

anubhav-dogra, 2019-01-18:
Hey Alex, 
Thanks it worked fine, and I made a config file.
The virtual (demo) is working fine. 

But as in picture below, some of the launch files are not created ! and I think that is why I am not able connect my Robot to moveit Rviz.
How to get those missing files ?

/home/anubhav/Pictures/Screenshot from 2019-01-19 01:18:33.png

alexvannobel, 2019-01-18:
Hi @anubhav-dogra ,

Those files were not autogenerated by the MoveIt Setup Assistant, but rather written by us. Our bringup nodes use the "**robot_model_demo**.launch" format, where **robot_model** is specified as a launch file argument most of the time. This is why we wrote those launch files. 
I think you should copy the missing files from the m1n6s300_moveit_config package into your own m1n4s200_moveit_config package and **modify** them to correctly reference you m1n4s200 configuration files (and not the m1n6s300 ones!)
I understand this is a bit cumbersome, but it has to be done in order to create a functional moveit_config package that works with our other launch files. 

Hope this helps, 

Best, 
Alex

anubhav-dogra, 2019-04-14:
hi @alexvannobel 

yes after a long time, I tried this and edited all those files.  
Robot is now connected with moveit and Rviz. but this time while executing a path I'm getting this error. Plz make understand and rectify about this, 

[ERROR] [1555234796.420174358]: Unable to identify any set of controllers that can actuate the specified joints: [ m1n4s200_joint_1 m1n4s200_joint_2 m1n4s200_joint_3 m1n4s200_joint_4 m1n4s200_joint_finger_1 m1n4s200_joint_finger_2 ]
[ERROR] [1555234796.420184942]: Known controllers and their joints:

anubhav-dogra, 2021-01-17:
Hi, 
I forgot to mention, that I was able to control mico 4 dof with moveit. Planning and execution is working. But the issue is speed of rotation of each joint is very slow, and thus arm is unable to reach its desired position. It will only after executing same plan multiple times. 
I have noticed that, when publishing the position of joints, the motion is smooth and fast. but when publishing velocity commands, the motion is very slow.  And I'm not able to find, where to change, to increase the speed and for smoother execution.

felixmaisonneuve, 2021-06-02:
Hi @anubhav-dogra 
What commands are you doing to get such results?
Are you publish velocity commands at a 100Hz rate?

anubhav-dogra, 2021-06-03:
Hi Felix
Thank you for your reply. I understood the problem about the unexpected motion: that was due to using virtual machine for ROS control. 
Using ubuntu platform itself as a host system, solved the problem. 
But I have one query regarding the output velocity feedback from ros, that why these velocity values are not smooth? (attached figure). Is it due to PID tuning? if yes, where to change the pid values inside the kinova-ros package? 
 
[image]

felixmaisonneuve, 2021-06-04:
Can you give more information on these data. How where they generated?

I tried publishing on topic `/j2n6s300_driver/in/joint_velocity` and reading the output of the topic `/j2n6s300_driver/out/joint_state` and values were constant. 

I also tried using MoveIt to move my arm, which also gave very constant values of veocity (I used the rqt autogenerated graph to produce these images): 
[image]
[image]

anubhav-dogra, 2021-06-08:
I generated this data using a python script, by recording the values from the related topics and then saving it in a text file. 
I am getting the  ``` m1n4s200_driver/out/joint_state''' data in radians, thus it cannot be compared on the same graph using rqt tools. 
here is the screenshot of the output of joint velocity and joint torque. 

``` m1n4s200_driver/in/joint_velocity ''' will always be smooth and straight lines because these are input commands and are given according to the trapezoidal velocity commands. 

![data](https://user-images.githubusercontent.com/44242975/121133206-b172e280-c84f-11eb-88cc-02d5eee6c434.png)

felixmaisonneuve, 2021-06-08:
You are right, `/m1n4s200_driver/in/joint_velocity` will always be smooth, my mistake.

I did some digging about the output values received. The values comes directly from the readings of the firmware of the arm. For example, joint 3 firmware will measure it is moving at a certain speed and that speed is passed all the way back to the ROS API. 
Unstable values that are received are not caused by a bug in the kinova-ros package. 

This is probably related to hardware restrictions (inconsistent friction on joint, inconsistent current, worn out components...) rather than software. I think it would be very hard (if not impossible) to change this type of behaviour. 

Why do you need those values to be smooth? It could be possible to manipulate the output data to draw a smoother curve, but it wouldn't change the received values directly.

anubhav-dogra, 2021-06-08:
Ok thanks Felix, 
I just wanted to know if there is any role of PID values for this type of ouput. And I also agree to your point about hardware restrictions. 
How can I tune the PIDs in ROS package?  
Values of the output torques are giving a large error too. Could there be any issues with the torque sensor in these actuators?

felixmaisonneuve, 2021-06-09:
Hi @anubhav-dogra,
I did not know about servoing and PIDs, so I didn't understand your question, I am sorry about that. 
You are right, changing the PID (if done correctly) could possibly reduce this type of output. 

Kinova does not provide a tool to tune the PIDs, but you can modify the values.
However, **it is not recommended** to do so because it could severly damage your robot and it would not be covered by your garanty as stated [here](https://www.kinovarobotics.com/sites/default/files/UG-008_KINOVA_Software_development_kit-User_guide_EN_R02%20%281%29.pdf) (pages 23-24) : 
> Setting these parameters out of the suggested range can
severely damage the robot and is not covered by the
> warranty.
 
Last week, you asked how to change the PID, the easiest way to do it would be by using the Development Center app in the *Advanced Settings* section.

You could also use the kinova-ros API. 
If you look in `kinova_driver/src/kinova_api.cpp`, you will find `setActuatorPID` and `setActuatorPIDFilter`. 
You could then create a function in `kinova_driver/src/kinova_comm.cpp` that would look something like this : 
```c++
/**
  *@brief Example on how to change PID
 */
void KinovaComm::setActuatorPID()
{
    boost::recursive_mutex::scoped_lock lock(api_mutex_);
    int result;
    int actuator_address[] = {16,17,18,19,20,21,25};
    float gainKp = 0.0f; // Suggested range [0,2]
    float gainKi = 0.0f;
    float gainKd = 0.0f; // Suggested range [0,0.1]
    for (int i=0;i<num_joints_;i++)
    {
        result = kinova_api_.setActuatorPID(actuator_address[i], gainKp, gainKi, gainKd);
        if (result != NO_ERROR_KINOVA)
        {
            throw KinovaCommException("Could not change PID", result);
        }
    }
    ROS_WARN("PID for all joints set to zero");
}
```
This would allow you to create your own service in `kinova_driver/src/kinova_arm.cpp` with something like this : 
```c++
set_actuator_pid_service_ = node_handle_.advertiseService("in/set_actuator_pid", &KinovaArm::setActuatorPIDCallback, this);

bool KinovaArm::setActuatorPIDCallback(kinova_msgs::setActuatorPID::Request &req, kinova_msgs::setActuatorPID::Response &res)
{
    kinova_comm_.setActuatorPID();
    return true;
}
```
Of course you would also need to modify the corresponding `files.h` and to create a custom ros service message (or you could use an existing one like `ZeroTorques` because you dont really need to use the message arguments).
I have not tried this solution (and do not plan to), so I have no garanty that this actually work (but it should I think). 

Hope this helps,
Felix

felixmaisonneuve, 2021-08-20:
Since this thread seems to be inactive, I will close it.

If you still encounter this issue, comment / re-open it or create a new issue.

Regards,
Felix

closed 2021-08-20 by felixmaisonneuve