Kinovarobotics/kinova-ros #203: KinovaCommException: Could not get the angular command (return code: 1015)

kracon7, 2019-01-07:
I'm using kinova-ros on my:
6-DOF Kinova-Mico2, 
Ubuntu 16.04 
and ROS-Kinetic. 
I'm connecting the arm and my laptop by USB cable.
I tried to bring up the arm in ROS but it kept throwing the following error message. I don't know how to solve this issue.

```
jc@jc-XPS-15:~$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s200
... logging to /home/jc/.ros/log/f8572fb6-12c7-11e9-b646-9cb6d0bcf2df/roslaunch-jc-XPS-15-10825.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
deprecated: xacro tags should be prepended with 'xacro' xml namespace.
Use the following script to fix incorrect usage:
        find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
when processing file: /home/jc/catkin_ws/src/kinova-ros/kinova_description/urdf/m1n6s200_standalone.xacro

xacro.py is deprecated; please use xacro instead
started roslaunch server http://jc-XPS-15:39113/

SUMMARY
========

PARAMETERS
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /m1n6s200_driver/connection_type: USB
 * /m1n6s200_driver/ethernet/local_broadcast_port: 25025
 * /m1n6s200_driver/ethernet/local_cmd_port: 25000
 * /m1n6s200_driver/ethernet/local_machine_IP: 192.168.100.100
 * /m1n6s200_driver/ethernet/subnet_mask: 255.255.255.0
 * /m1n6s200_driver/jointSpeedLimitParameter1: 10
 * /m1n6s200_driver/jointSpeedLimitParameter2: 20
 * /m1n6s200_driver/robot_name: m1n6s200
 * /m1n6s200_driver/robot_type: m1n6s200
 * /m1n6s200_driver/serial_number: not_set
 * /m1n6s200_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /m1n6s200_driver/torque_parameters/use_estimated_COM_parameters: False
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    m1n6s200_driver (kinova_driver/kinova_arm_driver)
    m1n6s200_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [10838]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8572fb6-12c7-11e9-b646-9cb6d0bcf2df
process[rosout-1]: started with pid [10851]
started core service [/rosout]
process[m1n6s200_driver-2]: started with pid [10868]
process[m1n6s200_state_publisher-3]: started with pid [10869]
[ INFO] [1546898569.623991012]: kinova_robotType is m1n6s200.
[ INFO] [1546898569.624385458]: kinova_robotName is m1n6s200.
[ INFO] [1546898569.627400844]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1546898569.782836601]: Found 1 device(s), using device at index 0 (model: Mico Assistive, serial number: Not Set, code version: 328960, code revision: 5)
[ INFO] [1546898569.855802690]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1546898569.999998555]: The arm is ready to use.
m1n6s200_joint_1 m1n6s200_joint_2 m1n6s200_joint_3 m1n6s200_joint_4 m1n6s200_joint_5 m1n6s200_joint_6 
terminate called after throwing an instance of 'kinova::KinovaCommException'
  what():  KinovaCommException: Could not get the angular command (return code: 1015)

[m1n6s200_driver-2] process has died [pid 10868, exit code -6, cmd /home/jc/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver m1n6s200 __name:=m1n6s200_driver __log:=/home/jc/.ros/log/f8572fb6-12c7-11e9-b646-9cb6d0bcf2df/m1n6s200_driver-2.log].
log file: /home/jc/.ros/log/f8572fb6-12c7-11e9-b646-9cb6d0bcf2df/m1n6s200_driver-2*.log

```

Also I tried the SDK for MICO2, I should see the serial number on the right conner according to this post [https://github.com/Kinovarobotics/kinova-ros/issues/95#issuecomment-309017810].
But here's what I see in the Dev center.
[image]

sbrisebois, 2019-01-08:
Hi,

I want to confirm with you if you use the master branch of the repo ? Because the master branch is currently for ROS Indigo. So it may cause instability.

Sylvain

kracon7, 2019-01-08:
Hi,
Here's what I got from git status in the wrokspace folder:

jc@jc-XPS-15:~/catkin_ws/src/kinova-ros$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Jiacheng

sbrisebois, 2019-01-08:
Ok, so definitively, this might be an issue.

Kinova does not officially support ros package for Jaco2 on kinetic.
But a collaborator develop a kinetic version of ros package.
Pull the master branch, then switch to commit 

https://github.com/Kinovarobotics/kinova-ros/commit/94c25e19443acb4fadac34c3f796f108f6a8471b
.
The steps should look like :
cd ~/catkin_ws/src
git clone https://github.com/Kinovarobotics/kinova-ros.git kinova-ros
cd kinova-ros
git checkout master 
git checkout 94c25e1
cd ~/catkin_ws
catkin_make
It works fine on our side, please let us know if it does not work

Sincerely,
Sylvain

kracon7, 2019-01-08:
It worked. Thanks!

Jiacheng

closed 2019-01-13 by nsaif-kinova