Kinovarobotics/kinova-ros #204: Incorrect inertial parameters for wrist (mesh_no-4) causes jitter for j2n6s300 in Gazebo

skasewa, 2019-01-11:
Please note that in `kinova_inertial.xacro` , the inertial parameters for wrist (mesh_no-4) aren't equal. 

```
  <!-- wrist = 4 -->
  <xacro:unless value="${mesh_no-4}">
    <inertial>
        <mass value="0.426367" />
        <origin xyz="0 -0.037 -0.0642" />
        <xacro:inertia_cylinder height="0.02"  radius = "0.04" mass="0.1785"/> 
    </inertial>  
  </xacro:unless>  
```

This causes the arm to jitter when simulated in Gazebo for model j2n6s300. To correct it, simply replace the bottom mass number with the top one.

 ```
<!-- wrist = 4 -->
  <xacro:unless value="${mesh_no-4}">
    <inertial>
        <mass value="0.426367" />
        <origin xyz="0 -0.037 -0.0642" />
        <xacro:inertia_cylinder height="0.02"  radius = "0.04" mass="0.426367"/>   <!-- mass="0.1785"/>  -->
    </inertial>  
  </xacro:unless>
```  

I would have submitted a pull-request, but this seems very tiny as a change to create a pull-request for.

N.B.: Not tested for other models.

skasewa, 2019-01-11:
Additionally, the contact with the ground plane also contributes to the jitter. After instituting the above fix, model still jitters, but stops after deleting the ground plane. Any fix for the ground plane issue would be appreciated, happy to open a separate issue for that.

nsaif-kinova, 2019-01-15:
Thank you Skasewa for your help, I will also transmit your request to our engineering department.
Sincerely,

skasewa, 2019-02-01:
Upon further investigation, there's a couple of more possible problems with the current xacros (for j2n6s300) : 

- (refer to image) Inertias of base and shoulder do not seem to be correct. They overlap, and the inertial box for the base sticks out above it, while that of the shoulder dips far below it. Are these how they should be? Is it possible that both links account for the actuator? Which of these would it be correct to have the actuator in?
- The correction in the mesh_no-4 mass value no longer eliminates the jitter
- kinova_inertial.xacro uses cylindrical approximations instead of values obtained from the real robot. I would hope for something more accurate from the official repository.

[image]

nsaif-kinova, 2019-02-04:
Hello Sudhanshu,
Are you using a modified version of the URDF files or configuration?
Do you have this problem happening if you download our ROS package from the branch master without any modification?
Sincerely,

skasewa, 2019-02-06:
Hey,

The only change I had made was the one mentioned in my first comment in this thread.

I made that change after observing and investigating the model jittering. However, I just tried again with the original xacro as you recommended, and right now it has run without jittering for 3 trials. It is therefore possible that there is nothing wrong with the xacro after all, and I'm grateful for your time in investigating this with me.

I do not know what caused the jittering in the first place though, and I hope it doesn't return.

Thanks.

HappySamuel, 2020-09-25:
Hi @skasewa 

Can you share your kinova_inertial.xacro? And also your j2n6s300_control.yaml

I also experience the jittering issue with the j2s6s200 model alone, so wish to use your parameters setting as a reference. 
But when i use it with mobile robot, the arm directly explode. #316 

Best,
Samuel

felixmaisonneuve, 2021-06-04:
To answer HappySamuel, skasewa solved his jittering problem by using xacro files available in the master branch.
I will therefore close  this issue.
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one.

Thank you

closed 2021-06-04 by felixmaisonneuve