Kinovarobotics/kinova-ros #205: Cartesian Velocity Control does not work (j2s7s300)

Kukanani, 2019-01-11:
We are attempting to control a j2s7s300 using cartesian velocity commands as described in the README, but the arm is not moving when we run the following commands:
```
# terminal 1
roslaunch kinova_bringup kinova_robot.launch kinova_robotName:=j2s7s300 kinova_robotType:=j2s7s300
# terminal 2
rostopic pub -r 100 /j2s7s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 1.0, twist_linear_z: 0.0, twist_angular_x: 0.0,
  twist_angular_y: 0.0, twist_angular_z: 0.0}"
```
The arm does not move. We can send joint velocity commands and the robot will move, but not cartesian. The only thing I can think of that is different from a "default" case is that we are using a Robotiq 2-finger gripper instead of the Kinova gripper. Other than that, we are using the latest `kinova-ros-beta` branch with no modifications. We see the same behavior on the `master` branch as well.

Kukanani, 2019-01-14:
OK, this makes sense. I was able to get motion in other areas. Thank you.

However, I'm now having trouble getting the arm to move below certain speeds. `twist_linear_z: 0.03` results in motion, but `twist_linear_z: 0.02` results in no motion. The same is true for `twist_angular_z: 0.2` and twist_angular_z: 0.1`.

nsaif-kinova, 2019-01-16:
Hello Kukanani,
Please note the arm apply some filters to the noise and with the PID we use sometimes the very small commands can get ignored, on my side I did a test with twist_linear_z: 0.02 and i was able to move the arm in the Z axis, same for twist_angular_z: 0.2, I can see the gripper rotating
Echoing joint angles can also help you verify if the arm is moving

Kukanani, 2019-01-18:
Hello Nassif,

Thank you for the response. I am still unable to move the arm slower than 0.03 in any linear direction. Is there any way to know whether or not a command is too small and will get ignored? Is this also dependent on robot position? Finally, is there any way to disable that behavior? We would love to be able to move the robot quite slowly (1cm/sec or less).

nsaif-kinova, 2019-01-21:
Hello Kukanani,
The lowest velocity speed I am able to achieve is 2 cm/s, however I am able to send position commands of 1 cm or even 0.5 cm, you can use this to move the arm at small steps, so you can achieve a 1 cm/s of average velocity  between two distant positions, since the controller frequency is 100 Hz you should send a loop of Cartesian positions with the value of 0.0001 or 0.00005 each (0.0001m*100Hz=0.01m/s) (please note the instantaneous velocity will not be maintained at 1 cm/s, the arm will stop before receiving the next command, but the average velocity will be 1 cm/s)
Sincerely,

Kukanani, 2019-01-22:
Thank you for the information. Unfortunately the solution you describe doesn't fit my use case, which is to move at a constant, slow speed. However, since there seem to be no lower bounds for when specifying joint velocities directly, I have been able to construct my own jogger using PyKDL that fits my needs.

Thanks again for the help!

closed 2019-01-22 by Kukanani