Kinovarobotics/kinova-ros #206: Compiling on arm64

Bennyhwanggggg, 2019-01-15:
Hello,

We are trying to compile our app on an ARM (aarch64) device. We tried to compile this repo on our machine, but hit an error on kinova)driver where ".../lib/aarch64-linux-gnu/USBCommLayerUbuntu.so" cannot be found

Would it be possible to include an ARM (aarch64) version? We can probably also compile it from source on our hardware.

Thanks!

nsaif-kinova, 2019-01-15:
Hi,
We are working to fix this problem.
Sincerely,

LazyEngineerToBe, 2019-08-30:
@nsaif-kinova Any update on this issue?

nsaif-kinova, 2019-09-03:
Hello Alexandre,
I am adding our support team to assist you @KinovaCustomerSupport 
Thanks

shelderzhang, 2021-03-18:
hi,
    @nsaif-kinova  Has this problem beed fixed? I hope to use this package on an ARM (aarch64) device, recently

felixmaisonneuve, 2021-06-07:
Hi,
We do not officially support ARM for our robot. 
However, some work has been done about this here : https://github.com/Kinovarobotics/kinova_sdk_recompiled/tree/master/raspberry%20pi%203

You will need to replace all the `kinova-ros/kinova_driver/lib/i386-linux-gnu/*.so` file (in your workspace) by the corresponding ones in the repo. 

This being said, there is no garanty this is gonna work.

I am gonna close this issue since we do not really support this feature and I cannot provide much help regarding this (I am not an expert at all on the subject).

Hope this helps

closed 2021-06-07 by felixmaisonneuve