Kinovarobotics/kinova-ros #209: error with run kinovaj2s7s300 in gazebo

wangqiaoli, 2019-01-29:
I have fatal problems when running the arm model in gazebo. 
system: ubuntu14.04 + indigo + gazebo2.2.3
robot:  a 7dof robot with spherical wrist.
the code: roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s7s300
the results:
[image]

nsaif-kinova, 2019-01-29:
Hello sorry I am not Gazebo expert, but I am able to launch this command without any problem :` roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s7s300`
Also why do I see at line 1548759062.593241973 j2n6s300
I guess something is wrong with your ROS package, can you please try to reinstall the ROS package?
Sincerely,

wangqiaoli, 2019-01-31:
> Hello sorry I am not Gazebo expert, but I am able to launch this command without any problem :` roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s7s300`
> Also why do I see at line [...]

Sorry. I have a mistake in my question. I've correct it now.  
No mater I set the konova_robotType:=j2s7s300 or set the konova_robotType:=j2n6s300, the errors occurred .
Could it be something wrong other than the ROS package?

wangqiaoli, 2019-01-31:
> Sorry. I have a mistake in my question. I've correct it now.
> No mater I set the konova_robotType:=j2s7s300 or set the konova_robotType:=j2n6s300, the errors occurred .
> Could it be something wrong [...]

what's your platform configuration?  which versions of ubuntu, ros and gazebo do you use?

nsaif-kinova, 2019-02-01:
Hello wangqiaoli,
I am not Gazebo expert, I can confirm that following this guide https://github.com/Kinovarobotics/kinova-ros/wiki/Gazebo on my computer I can simulate our arms in Gazebo,
Did you install these packages?
```
sudo apt-get install ros-indigo-gazebo-ros-control
sudo apt-get install ros-indigo-ros-controllers*
```
Sincerely,

sbrisebois, 2019-02-02:
Hi Wangqiaoli,

One thing that may explain this, is the type of computer you are using. Gazebo is very demanding on GPU and CPU power, and on my laptop it doesn't run well, it crash often. If I run the same repo on a sysytem with a good video card, and a good cpu, it run perfecty.

This may be caused by the delay that require each component to load.

alecGraves, 2019-03-15:
Installing the ros-controllers solved this issue for me (for anyone else who happens by this page)

rebecatourinho, 2019-04-11:
I've tried the kinetic branch on Ubuntu 16.04 and got same error - Unexpected exception caught when initializing trajectory from ROS message data - even after setting up legacyModeNS true according to this issue i have had too https://answers.ros.org/question/292444/gazebo_ros_control-plugin-gazeboroscontrolplugin-missing-legacymodens-defaultrobothwsim/.

nsaif-kinova, 2019-04-12:
Hello,
Can you please tell me the specs of your computer? (CPU, RAM, Graphic card...)

closed 2021-06-04 by felixmaisonneuve