Kinovarobotics/kinova-ros #212: Jaco MoveIt does not receive valid timestamp

philzook58, 2019-02-05:
In MoveIt, the jaco arm is not executing plans. It fails with the message

```
[ INFO] [1549397138.885220314]: Didn't received robot state (joint angles) with recent timestamp within 1 seconds.
Check clock synchronization if your are running ROS across multiple machines!
[ WARN] [1549397138.885519910]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[ INFO] [1549397138.885640144]: Execution completed: ABORTED
```

When I look at the joint state topic, it does appear to have an invalid timestamp that is not changing
```
---
header: 
  seq: 11384
  stamp: 
    secs: 1133099831
    nsecs: 1126656057
  frame_id: ''
name: [j2n6s300_joint_1, j2n6s300_joint_2, j2n6s300_joint_3, j2n6s300_joint_4, j2n6s300_joint_5,
  j2n6s300_joint_6, j2n6s300_joint_finger_1, j2n6s300_joint_finger_2, j2n6s300_joint_finger_3]
position: [4.809861741546026, 2.9223372642046597, 1.0012534950188776, 4.203945189285595, 1.4454578458836254, 1.323009161071199, 0.0, 0.0, 0.0]
velocity: [-0.007791646811650356, -4.891454401121671e-47, -4.891454401121671e-47, -0.011333306723049174, -4.891454401121671e-47, -0.011333306723049174, 0.0, 0.0, 0.0]
effort: [0.0, 0.0, 4.718234062194824, -2.5625240802764893, 0.0, -2.202988862991333, 0.0, 0.0, 0.0]
---
```

MoveIt does accurately show the correct position of the arm when moved with the joystick.

Ubuntu 16.04
```
roslaunch j2n6s300_moveit_config j2n6s300_demo.launch
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:="j2n6s300"
```
Any suggestions?

Thanks,
Phil

nsaif-kinova, 2019-02-05:
Hello Philip,
The kinetic package is not officially supported and tested by Kinova.
I had a similar error in the past while working with the kinetic package and it was due to an update on my Ubuntu 16, I did recompile my package and everything worked fine.
Can you please try to do catkin_make on your package?

philzook58, 2019-02-06:
On deleting everything and doing a fresh build, it seems to have worked. Thanks!

closed 2019-02-06 by philzook58