Kinovarobotics/kinova-ros #213: Weird invalid goal in moveit

jingxixu, 2019-02-12:
I am using m1n6s200 to grasp a block. MoveIt! reports "Unable to sample any valid states for goal tree" when I try to manually set a goal using the interactive marker which is shown in the following screenshots. There is clearly no collision but I do not understand why the goal is invalid. I have spent many days on this and cannot figure out why. A good grasp needs to be very close to the target so that I have to move the arm there but MoveIt is preventing me from doing that. Can you guys help? Thanks!

[image]
[image]
[image]

nsaif-kinova, 2019-02-12:
Hello Jingxi,
Can you please tell me what version of ROS you are using? is it Kinetic or Indigo?
Currently we officially support ROS Indigo
Sincerely,

jingxixu, 2019-02-12:
Hi, 

I am using ROS Kinetic. Do you think it is a ROS version problem? Is it possible to have some quick fixes?

Thanks!

nsaif-kinova, 2019-02-15:
Hello Jingxi,
Sorry for the delay, Yes this might be the cause of your problem since the kinetic package was not officially developed and tested by Kinova.
Can you please try with ROS Indigo? On my side moveit with Indigo is working fine.
Sincerely,

closed 2021-06-04 by felixmaisonneuve