Kinovarobotics/kinova-ros #216: Kinova_driver using raspberry pi (ubuntu 16.04)

joekeo, 2019-02-28:
Hello, 
Is there any way to make the kinova_ros work with the armhf architecture of the raspberry pi 3 B+?
I have tried to make it work with no sucess. Any pointers?

PD: I have tried adding the files from
https://github.com/Kinovarobotics/kinova_sdk_recompiled
onto the kinova_driver folder

sbrisebois, 2019-02-28:
Hi,

I just have test the sdk_recompiled and yes I confirm that it's not working on the pi3.
We have start the work to have a new compiled version that will run on the PI3.

As soon as we have a version I will notice you.
Also, we will update the github repo with the correct api build.

Sincerely,
Sylvain

joekeo, 2019-02-28:
Many thanks for your help,

I thought I was going crazy :)

I will be very happy to have the Kinova arm working with a Pi.

If there is anything I can help on, let me know.

Best,
Pepe

sbrisebois, 2019-03-01:
Hi Pepe,

I have just complete the re-build of the kinova API library for the PI3 UbuntuMate
It's updated on the github repo you had try.
https://github.com/Kinovarobotics/kinova_sdk_recompiled

I recommand that you try first the new lib with your arm and using a simple c example (not in ROS).
If this work (it sould be, because I try it myself...) then update the .SO files in the ros package with the new one from the repo.

I want also to give you a quick hint for ROS on the PI3 : 
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi

I havn't try the ROS yet, but I will do as soon as possible. In the mean time, you might try it by yourself and don't hesitate to write to us again if it's not working.

Best,
Sylvain

sbrisebois, 2019-03-01:
Hi Pepe,

I just realise that the instruction was for Raspibian. Here is the instruction for Xenial : 
http://wiki.ros.org/ShadowRepository

joekeo, 2019-03-07:
Hello Sylvain,

Many thanks for your help. Sorry for being silent, but I am back. 

I will receive the Kinova arm next week (as it was sent to another place for a bit) so I will be able to test the functionality with the Pi. I will let you know how it goes.

I appreciate a lot your help and quick response.

Best,
Pepe

sbrisebois, 2019-03-12:
Hi Pepe,

I have a good news, I'm currently setting-up a Pi with ROS and our package
already install on it.
As soon as it's working, I will put an image on the web to help you
accelerate the developpment.

Sincerely,
Sylvain

Sylvain Brisebois Ing.

Expert d'application client - division Innovation

Field application expert - Innovation division

514-277-3777 ,4078

514-244-8238

sbrisebois@kinova.ca

4333 boul. de la Grande-Allée, Boisbriand (QC)
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

<https://maps.google.com/?q=4333+boul.+de+la+Grande-All%C3%A9e,+Boisbriand+(QC)++%C2%A0+J7H+1M7&entry=gmail&source=g>J7H
1M7
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

kinovarobotics.com <http://www.kinovarobotics.com/>   Facebook
<http://facebook.com/kinovarobotics/>  Twitter
<http://twitter.com/kinovarobotics>  LinkedIn
<https://www.linkedin.com/company/kinova/>  YouTube
<https://www.youtube.com/user/Kinovarobotics>

The information contained in this email message is private and
confidential. It is intended for the exclusive use by the recipient named
above. If you are not the intended recipient, be advised that disclosing,
copying, distributing or taking any action in reliance on the contents of
this information is strictly prohibited. If you have been sent this email
by error, please destroy it and immediately inform us at
info@kinovarobotics.com.

joekeo, 2019-03-13:
Hello Sylvain,

This is amazing! It is great news. I will gladly be a test subject for that! 
We get our Kinova arm next week, then I can start playing with it again.
Please let me know when you have the image :)

Best,
Pepe

burakaksoy, 2019-06-12:
Hi, 

I have the same issue. Is there any progress on setting-up a Pi with ROS for Kinova Arm?

Thanks,
Burak

Devin521314, 2020-04-24:
Hello,
Is there any way to make the kinova_ros work with the armhf architecture of the raspberry pi 3 B?
I have tried to make it work with no sucess. Any pointers?

PD: I have tried adding the files from
https://github.com/Kinovarobotics/kinova_sdk_recompiled
onto the kinova_driver folder
Please let me know when you make it successfully.

closed 2021-06-04 by felixmaisonneuve