Kinovarobotics/kinova-ros #217: The finger_1 of the 3finger gripper is not correctly assembled.

xuhuairuogu, 2019-03-06:
The finger_1 is wrongly assembled. Maybe the orientation should be modified.
[image]

nsaif-kinova, 2019-03-06:
Hi,
Can you explain a bit more, what do you mean by wrongly assembled, which one is the finger 1?
Did you use our URDF files?
Sincerely,

xuhuairuogu, 2019-03-06:
Yeah. I just used the kinova-ros package. I have display the j2n6s300 model in rviz, gazebo and vrep, all of which shows that the axis of finger 1(the right one in the above picture) does not align with the axis of the palm.

nsaif-kinova, 2019-03-06:
Hello xuhuairuogu,
Can you please send me a picture of the real gripper highlighting the error you see?
Sincerely,

xuhuairuogu, 2019-03-06:
I do not have a real robot at hand. All is based on the observation in simulation. I think the value of the finger 1's rotation parameters  should be modified. My environment is Ubuntu16.04 + ROS KInetic + gazebo 7.14.0 + vrep 3.6.0

xuhuairuogu, 2019-03-06:
[image]

nsaif-kinova, 2019-03-06:
Hello xuhuairuogu,
We officially support ROS indigo branch for Gen2, the URDF model of the fingers I believe is correct, but I will take in consideration your comments.
For more information about our Gen2 please refer to this link https://www.kinovarobotics.com/en/knowledge-hub/gen2-ultra-lightweight-robot
where you can find the user's guide, the CAD model and many more information.
Sincerely,

closed 2021-06-04 by felixmaisonneuve