Kinovarobotics/kinova-ros #218: KinovaCommException: Could not get general information about the device (return code: 1015)

joekeo, 2019-03-07:
Hello,

I am testing the Kinova (j2n6s300) with a Intel NUC, running 16.04. 
I have previously been able to fully control the arm with this setup, same software, same hardware connections and such. In the past every other time trying to connect to the arm I would have the eror code (1015) when running the kinova_bringup launch file. But this time it has been all evening and I haven't been able to connect more than once.

When I plug in the joystick directly to the arm I have not problem 'home-ing' it but the NUC is not able to connect to it. 

I am running this NUC headless, Is there anyway I can debug what is happening?

Sometimes it runs the bringup without errors, but gets stuck in the 'initializing the gripper' part and stays there.
Other times it full starts but when I test move it using the:
 rosrun kinova_demo joints_action_client.py -v -r j2n6s300 degree -- 0 0 0 0 0 0

the bringup crashes

Any pointers?

Best,
Pepe

PD: full error log

rossrobo@HLP:~$ roslaunch kinova_bringup kinova_robot.launch 
... logging to /home/rossrobo/.ros/log/a99b2b70-4102-11e9-9221-f44d306c94d5/roslaunch-HLP-4099.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://HLP:42769/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [4112]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a99b2b70-4102-11e9-9221-f44d306c94d5
process[rosout-1]: started with pid [4125]
started core service [/rosout]
process[j2n6s300_driver-2]: started with pid [4142]
process[j2n6s300_state_publisher-3]: started with pid [4143]
[ INFO] [1551981531.386799455]: kinova_robotType is j2n6s300.
[ INFO] [1551981531.387249212]: kinova_robotName is j2n6s300.
[ INFO] [1551981531.390831594]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ERROR] [1551981531.392403756]: KinovaCommException: Could not get general information about the device (return code: 1015)

[j2n6s300_driver-2] process has died [pid 4142, exit code -11, cmd /home/rossrobo/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6s300 __name:=j2n6s300_driver __log:=/home/rossrobo/.ros/log/a99b2b70-4102-11e9-9221-f44d306c94d5/j2n6s300_driver-2.log].
log file: /home/rossrobo/.ros/log/a99b2b70-4102-11e9-9221-f44d306c94d5/j2n6s300_driver-2*.log

joekeo, 2019-03-07:
I have the permissions for the usb devices set up correctly (as the readme says)
And only to reiterate, the arm has been run before in the current hardware/software configuration

nsaif-kinova, 2019-03-07:
Hello Pepe,
We officially support ROS indigo, ROS kinetic was developed by the community.
Are you using the branch master? if yes this branch is for Indigo.
Are you able to have a steady connections using the Dev Center?
Did you try to use another usb cable and port?
Error 1015 is a connection problem, please refer to these issues :
https://github.com/Kinovarobotics/kinova-ros/issues/203
https://github.com/Kinovarobotics/kinova-ros/issues/135
https://github.com/Kinovarobotics/kinova-ros/issues/104
Sincerely,

edsut, 2019-08-06:
Hi,
I have a similar issue.  I am running Ubuntu 16.04 with ROS/Kinetic, and I used the branch that is suggested in https://github.com/Kinovarobotics/kinova-ros/issues/203.
That made no difference for me; so I dug into the code, and have what *appears* to be a workaround (sure would like to confirm this with Kinova folks)...

So far I've modified three different failure points in kinova_comm.cpp..
The 1015 error is defined as ERROR_NO_DEVICE_FOUND.
I assume that is referring to the USB device that the ARM shows up as, so it doesn't really make
sense 'cause the device IS there and was working up to the point of the error.
Anyway, without digging any further I just put a few (10) retries in the code that detects the 1015 return error placing a sleep(1) between each retry.  
It didn't eliminate the error, but the retries (usually one or two) allow the code to continue running.

Can someone from Kinova confirm that this is an ok workaround?  This doesn't appear to have anything to do with Indigo vs Kinetic.  More likely its just a flakey USB interface that needs to be accounted for.

Thanks in advance,
Ed

edsut, 2019-08-06:
If anyone is interested in the changes I made, reply to this and I'll post the patch file.

codedeamon, 2020-04-16:
>If anyone is interested in the changes I made, reply to this and I'll post the patch file.
Hi @edsut , would you mind posteing the changes you made to the code? I am trying to launch the arm in gazebo simulation without any success. 
Thank you :)

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over a years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve