Kinovarobotics/kinova-ros #224: roslaunch kinova_bringup kinova_robot.launch failed

xuhuairuogu, 2019-04-02:
I have got a new Jaco these days. I have installed JACO2SDK on Ubuntu16.04 successfully, connected Jaco and Ubuntu by USB and it is OK to control the robot by pressing the buttons in the Angular Control tab.So, I think there is no problem with communication and driver. Then, I copied the 10-kinova-arm.rules file to /etc/udev/rules.d directory. However, when I try to control Jaco through kinova_ros package by the command 'roslaunch kinova_bringup kinova_robot.launch', something wrong goes like the following:
[image]

nsaif-kinova, 2019-04-02:
Hello,
Do you have the Dev center or Torque consol open when you try to launch ROS?
Please note Jaco2 can only connect to one interface, so Please close the Dev Center and Torque Console before you use ROS.
Sincerely,

xuhuairuogu, 2019-04-02:
Yeah, I close both before launch

xuhuairuogu, 2019-04-02:
It is OK after reboot Ubuntu

closed 2021-06-04 by felixmaisonneuve