Kinovarobotics/kinova-ros #227: MoveIt! gripper group execution frequently fails

alecGraves, 2019-04-05:
Using moveit to control the gripper position for the arm is frequently failing on the real robot, but it works flawlessly in gazebo.

For example, running this python script with 
```
roslaunch j2s7s300_moveit_config j2s7s300_gazebo_demo.launch
roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s7s300 is7dof:=true
``` 

works fine.
```python
# this_python_script.py
import rospy
import moveit_commander as mc

gripper_joints_closed = [1.0, 1.0, 1.0]
gripper_joints_open = [0.1, 0.1, 0.1]

mc.roscpp_initialize(sys.argv)
rospy.init_node('pick_place_ar', anonymous=True)

gripper_group = mc.MoveGroupCommander("gripper")
gripper_group.set_joint_value_target(gripper_joints_closed)
plan = gripper_group.plan()
gripper_group.execute(plan, wait=True)

gripper_group.set_joint_value_target(gripper_joints_open)
plan = gripper_group.plan()
gripper_group.execute(plan, wait=True)

mc.roscpp_shutdown()

```

But running this same script with the actual robot:
```
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300
roslaunch j2s7s300_moveit_config j2s7s300_demo.launch
```

often fails to move the fingers at all. I end up rebooting the robot several times and eventually it will get into a state where it can respond to gripper commands (I would say maybe 1 in 5 reboots).

In the moveit window, I am getting the following messages:
```
[ INFO] [1554487673.459507200]: Execution request received
[ERROR] [1554487675.854087700]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 2.297810 seconds). Stopping trajectory.
[ INFO] [1554487675.871658200]: Cancelling execution for j2s7s300_gripper
[ INFO] [1554487675.880738400]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1554487675.884367500]: Execution completed: TIMED_OUT
```

The driver window seems fine, but I am occasionally getting:
```
[ WARN] [1554487716.709829000]: Hardware Notification:USB SCP overflow,1.55449e+12,Error,Hardware Error
[ WARN] [1554488064.247970700]: Hardware Notification:USB SCP overflow,1.55449e+12,Error,Hardware Error
```

The MoveIt! functionality for the arm group works fine, I can execute angle targets and ee pose targets without any errors. Also, I can use the driver's action server to move the fingers without problems. Is there anything that is known about how to get the fingers to move reliably with moveit?

It is much more convenient for me if the simulation has the same interface as the physical bot without needing a wrapper and having to toggle between action server and moveit.

sbrisebois, 2019-04-09:
Hi Alec,

We have reproduce your issue on our side.

I will raise a fix request to the engineering.

But... I have a workaround if it fit to you...
I tried the Master branch (was updated recently) on Ubuntu 16.04 (Ros Kinetic) and it works very well with your code.

So you may pass to Ubuntu 16 if this not break your current development.

Let me know if this help you.

alecGraves, 2019-04-14:
Hi Sylvain,

Thank you for your feedback, I am unable to switch to Ubuntu 16 for my current project, but thank you for the recommendation. I ended up checking ros if the driver node is running and using the action server to move the gripper.

It would be great if the engineering team can find the bug; I am looking forward to a solution 😃

flymaxty, 2020-07-21:
Hi,
I have the same problem when using gripper group.

Gripper execution would raise TIMEOUT states, if the gripper can not reach the target on time.
This will happen when you are trying to catch something, but the value you request is narrower than the object size.
Which means the gripper has already catch the object and stop, but not reach the pose you request.

I add a timeout to gripper_command_action_server.cpp,  to return SUCCEEDED states to the actionlib client when the timeout reaches. But, this might make it difficult to find gripper mechanical failure, which is another situation the gripper can not get to the request pose.

felixmaisonneuve, 2021-06-04:
This issue has been inactive for a couple months now, I will close it. 
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one. 

Thank you

closed 2021-06-04 by felixmaisonneuve