Kinovarobotics/kinova-ros #229: Use ros_control package to write custom hardware interface and controllers

ZhengYi0310, 2019-04-17:
Hi, I'm considering writing some custom hardware interface by warpping kinova_comm with [ros_control](http://wiki.ros.org/ros_control), the good side of using ros_control is that I can implement some custom controllers and use it in a more flexible way by [controller_manager](http://wiki.ros.org/controller_manager). Another thing is that combining ros_control and Xenomai kernel I can make the library for kinova real time, although since I'm using the high level api, which run at only 100hz, it might have less real time issues.

Any suggestion or thoughts on this ?

Thanks

nsaif-kinova, 2019-04-18:
Hello Zheng,
Using the High level API we can go up to 100Hz, to go faster the use of Low level API is needed. Please note the loop control of the actuators is at 1Khz, but the base run at 100Hz.
Sincerely,

ZhengYi0310, 2019-04-18:
Hi Nassif,
    To make my questions more clear, here are some following up questions.
    1. At what circumstances do you think going to low level API is
worthwhile instead of staying at high level API ?
    2. For the base, I assume you mean the DSP ? what's the consideration
of using a DSP, which can only communicate with PC at 100hz, while the
direct communication of actuators can go to 500hz, is it because of the
easiness of API development ?
    3. It looks like some parts in kinova_gazebo uses ros_control, why
doesn't kinova develop such a framework for the actual robot ? since it's
is more flexible and more accurate in terms of timing(because ROS is not
realtime safe) than the current solution, that the command will be
published to a topic and the subscriber in kinova_arm.cpp will take it and
call relevant kinova_comm functions. If instead of computing the whole
trajectory be-forehead and publish it to the corresponding command topic, I
want to compute some step-by-step low level feedback control commands and
send it to kinova (let's say at 100hz, and I want to cycling time to be
relatively accurate around 100hz), the the current solution in kionva_ros
might be inconvenient for that ?
    4. from another pose
https://github.com/Kinovarobotics/kinova-ros/issues/115, I read that if I'm
using position controller, then the behavior of the arm will be moving to
at every goal point and stop, that means if I have a custom controller send
goal command positions every 100hz, then the arm will move and stop every
100hz, can you confirm that ? And if this is the case, then it looks like I
need to send velocity commands or torque commands (which are computed using
PID) if I want to get a continuous trajectory, can you confirm that ?
    5. If I want to compute command torques using PID, do you have
suggestions in terms of proper gains for each joint, my robot is j2s7s300.

    Thanks

ZhengYi0310, 2019-04-18:
Also, according to [https://github.com/Kinovarobotics/kinova-ros/issues/141](https://github.com/Kinovarobotics/kinova-ros/issues/141), the staff says "the actuators include a PID controller on the position (used when doing position control... which you do in Moveit) and a P controller for the torque (used when doing torque control). The default gains are P=2.0, I=0, D=0.05.". So does this mean, if I have my current joint position from KinovaComm::getJointAngles, then I have some desired goal position for the joint, then I compute something using a PID with gains set as mentioned in the post, then what I get is actually not a torque, therefore I should not call KinovaComm::setJointTorques, but something that should be sent directly to the actuator using lowlevel API ?

Thanks

nsaif-kinova, 2019-04-18:
Hello Yi,
1- The high level API will manage all the safeties on the arm, and the inverse kinematic. The low level API give you the ability to send and receive commands at faster rate, but you will be responsible to do the inverse kinematic and send the right commands to not break the arm.

2-The base is the controller, please refer to the previous point, if you are willing to manage the inverse kinematics and the safety on the arm, you can use the low level API for a direct communication with the actuators.

 3-I am not sure to well understand your question, the ROS package is a wrapper around the C++ API, all commands are sent using C++, Gazebo is for simulation, you don't need a physical arm to use it.
For more information about Gazebo please refer to this link : https://github.com/Kinovarobotics/kinova-ros/wiki/Gazebo
You can implement your own control loop, or you can use the Kinova inverse kinematics inside the controller when the robot receives a Cartesian trajectory point (or Cartesian position), it computes a straight line from its current position to the next trajectory point. Then it computes the next reachable point on this straight line considering maximum velocity and acceleration limits and gets deviated by its automatic self collision and singularity avoidance algorithms.

4-When you send multiple position commands the arm will go to your points one by one, the velocity will go down to 0 before it goes to the next point. If your own control loop send velocity commands then the arm will not stop, this is the case of moveit controlling the arm.

5-Using the high level API you can send torque commands, do you want to use the low level API for torque?

If you send joint positions to the actuators, each one has its own control loop and PID which will take care of reaching that position, the control loop is running at 1Khz.
You can not use position and torque commands at the same time, you have to switch from one mode to another.

Please let me know if you have more questions.

closed 2019-04-23 by ZhengYi0310