Kinovarobotics/kinova-ros #230: get tool cartesian velocity

xli4217, 2019-04-18:
Is there support for getting the tool frame linear and angular velocity? thanks.

nsaif-kinova, 2019-04-18:
Hello Xiao,
Please let me know if this will help you : https://github.com/Kinovarobotics/kinova-ros/issues/193#issuecomment-447343983

xli4217, 2019-04-19:
I didn't seem to find anything on getting end-effector velocities in #193. Maybe I'm missing something.

nsaif-kinova, 2019-04-23:
Hello Xiao,
This example is intended to apply a transformation to a joystick command to control the translation in the gripper frame instead of the base frame.
You may need to modify it to fit your needs, but you can refer to this example on how to apply the transformation, form the end effector velocity according the base to the tool pose.
Sincerely,

polthergeist, 2020-03-20:
Hi @xli4217, I will close this issue but feel free to reopen it if your problem was not solved or if you have other question regarding the Cartesian velocity.

closed 2020-03-20 by polthergeist