Kinovarobotics/kinova-ros #231: Ubuntu 14.04 to 16.04 upgrade

kurenaloha, 2019-04-19:
Supported versions are as follows in README.md.

### **Supported versions**
The recommended configuration is ROS Indigo with 64 bit Ubuntu 14.04.
The package may work with other configurations as well, but it has only been tested for the one recommended above.

However, the official support for Ubuntu 14.04 will be end at the end of April.
Which  version is better to move JACO?

alexvannobel, 2019-04-23:
Hi @kurenaloha ,

 -If you are more interested in Kinetic, you can take a look at issue #117 for more details. You will also find the link to a fork which was modified to work with Kinetic.
 - If you are more interested in Melodic, I invite you to check out #219 .
I personally would go with Melodic since support will last longer.

Hope this helps,
Cheers,
Alex

kurenaloha, 2019-04-24:
Thank you  for your reply. @alexvannobel 

I'll check out #117 and #219.
Thank you very much!

closed 2019-04-24 by kurenaloha