Kinovarobotics/kinova-ros #232: about jaco ik

xuhuairuogu, 2019-04-22:
Hi, 
  I have a Jaco2 with non spherical wrist and am working on its inverse kinematics. I just copy the jaco_ik.cpp file under kinova_driver/src/ikfast directory and ikfast.h file under openrave/python firectory and compile them successfully like the following:
[image]
 Then I test it with the configuration and pose given in the manual:
[image]
[image]
 The physical angles is [pi,3pi/2,pi/2,pi,pi,0] and the corresponding DH angles is [-pi,pi,pi,0,0,pi/2]. Substituting the DH angles into forward kinematics which used standard DH convention, I get the pose of the end effector T = [0 -1 0 0.41;
                                                                               0.5 0 -0.866 -0.2399;
                                                                               0.866 0 0.5 0.084;
                                                                               0 0 0 1];
However, the ik solutions given by the executable do not consistent with the pose:
[image]

nsaif-kinova, 2019-04-23:
Hello,
This file is for IKfast, did you also check this folder including the .cpp for IKfast Moveit?https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_moveit/kinova_arm_moveit_demo/src
For more information about IKfast please visit : http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html
Please note the controller do not use the IKfast, Kinova has its own inverse kinematic in the controller, the IKfast is only used for Moveit.
Please let me know if I can help you more.
Sincerely,

Hubert51, 2020-08-01:
Can we do inverse kinematic in gazebo simulation?

felixmaisonneuve, 2021-06-04:
This issue has been inactive for a couple months now, I will close it.
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one.

Thank you

closed 2021-06-04 by felixmaisonneuve