Kinovarobotics/kinova-ros #233: Very strange KinovaCommException instance

ZhengYi0310, 2019-04-23:
Hi,
   I'm using [ros_control](http://wiki.ros.org/ros_control) to rewrite the ros package, in order to implement my own control loop later, and in my [hardware_interface](http://wiki.ros.org/hardware_interface?distro=indigo), I implemented [a general read and write loop](http://wiki.ros.org/ros_control/Tutorials/Create%20your%20own%20hardware%20interface) use leveraging KinovaComm, and the following is the code snippet of the hardware interface:
[image]
[image]
[image]
[image]

in the following is the code snippet for the control loop:
[image]

Then, there is another node that will call the KinovaHWInterface::setTorqueControlModeService during my experiment (imaging I want to first set the arm to position control mode, let it run a trajectory, then set it to gravity compensation mode so I can move it back), I get the following exception:
[image], so the try catch block catches a communication exception during the process, and I wonder what is the reason of this, because everything looks normal...The loop is running at 100hz, my system is ubuntu 14.04 LTS, ros indigo, the kinova_ros package I'm using is also indigo branch.

Thanks

sbrisebois, 2019-04-23:
Hi Yi Zheng,

My first assumption wold be that you are calling multiple API command at the same time.
The Jaco2 API is not thread safe, and using services with API call may cause big issues.

The first thing to try is to add a mutex on the kinova_comm_ object.
Each time you use the object you must gain the mutex to avoid parallel call.

Let me know if this help.

Sylvain

ZhengYi0310, 2019-04-23:
Hi  sbrisebois,
     
Thanks for the replay !
     
When you say I'm calling multiple API command at the same time, do you mean commands like kinova_comm_->getJointAngles(), kinova_comm_->getJointVelocities(), kinova_comm_->setTorqueControlState() etc. ?
    
 If yes, then I'd like to clarify that the kinova_comm_ object is a boost::shared_ptr of kinova::KinovaComm declared in kinova-ros/kinova_driver/src/kinova_comm.cpp, which by construction, already has a internal boost::recursive mutex , and there is boost::recursive_mutex::scoped_lock lock(api_mutex_) been called inside functions like getJointAngles(), getJointVelocities(), etc.
   
 If you mean that I need to add another mutex for my control loop besides this, then it's a bit strange because I see in kinova-ros/kinova_driver/src/kinova_arm.cpp, there are also functions which will call kinova::KinovaComm function to get joint status and tool status without adding a mutex, such as KinovaArm::statusTimer(const ros::TimerEvent&).

Basically what I'm doing is just implementing my own kinova::KinovaArm class, without modifying anything in kinova::KinovaComm, but just leverage it.

Thanks

sbrisebois, 2019-04-23:
Hi Yi Zheng,

Good catch !. Please tell us if this fix your problem, we will fix it in the repo

Best,
Sylvain

ZhengYi0310, 2019-04-23:
Hi sbrisebois ,
    I did multiple tests, at least for my system, the problem appears to be gone.

    Thanks

sbrisebois, 2019-04-24:
Thanks,

I will raise a flag to our eng dept to fix this in the repo.

Many thanks for your inputs and help.

Best,
Sylvain

ZhengYi0310, 2019-05-02:
Hi sbrisebois,
    It looks like all the robot state reading functions not part of the high level api loop ? My general hardware interface loop is like this:
    
[image]

and my read function is like this:
[image]

and my write function is like this:
[image]

even if I commented out controller_manager_->update(ros::Time::now(), elapsed_time_), so bascially I'm doing nothing and all joint_velocity_command_ fields are 0, the cycle time of the loop can is 0.2 secs, as you can see in the following image:
[image]

So does this basically mean we should never call functions getting robot state and functions sending command of high level API in the same thread ?

Thanks

ZhengYi0310, 2019-05-02:
by commenting out the two lines in read function that get cartesian position and cartesian force, the cycle loop is 0.013 ~ 0.014 secs. But it's still giving me trouble, because when I uncomment the  controller_manager_->update(ros::Time::now(), elapsed_time_), I will have another thread computing desired velocities and assign the value to joint_velocity_command_, but since this loop is running lower than 100hz, the robot is unable to track the desired joint velocities.

KinovaCustomerSupport, 2019-05-13:
Hi ZhengYi0310,

I read through your code and for me the duration of the loop is partially explained by the number of API call you do each loop.

Each API call send a USB command to the robot (it runs at 100hz, it handles each message at 10ms) and most of the command expect a return message, so each API call take several ms to run since they are blocking.

You do 6 api calls in your read alone, it could take more than 100 ms.

What I would suggest is to split the execution of the calls to read only critical information and write it back to the arm each iteration. Also, read and write the rest every once in a while (1 to 10 iteration).

Let me know if it helps,
Jean-Philippe Côté
Customer support

closed 2019-06-04 by jpcotekinova