Kinovarobotics/kinova-ros #234: mico cannot move using moveit

wangxian4423, 2019-04-25:
Hi,I'd like to ask a question.While I`m using the mico j1n6s300 with the moveit,I find it cannot execute though the result shows Executed.It looks like this:
[image]
and I find the rqt_graph show this:
[image]
The /m1n6s300_joint_trajectory_action_server publish a topic called /m1n6s300_driver/trajectory_controller/command and it works well.Howerer,it seems like no node has deal with this topic,so the arm cannot be command.
So what can I do to solve this problem?
Thank you very much if you can help me~

RalphChahine, 2019-04-25:
Hi, I have tested mico on my part to try and reproduce the error. However, all worked fine for me. I was able to control the arm through moveit with no problems on both Ubuntu 14 and Ubuntu16.  Here are the commands that I used : 
1) roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s300
2) roslaunch m1n6s300_moveit_config m1n6s300_demo.launch

Is there an error that appears on the terminal when you do the command 1)?
Is there an error when you do the command 2)

If there is, please send me what the errors are so I can troubleshoot your issue

Ralph

wangxian4423, 2019-04-26:
Thanks for your reply,well,there is an error when I  do the command 2)
`[ERROR] [1556242694.061319794]: SpinnerMonitor: single-threaded spinner after multi-threaded one(s).Attempt to spin a callback queue from two spinners, one of them being single-threaded.
In the future this will throw an exception! Only allowed for backwards compatibility.
`
Besides,the joystick is Flashing green light like this and I cannot use the joystick to control the arm.
[image]
Though the arm can be control using the  GUI form the SDK.

RalphChahine, 2019-04-26:
You are facing a boot loader issue since you cannot control the arm with either the joystick or ROS and the green lights are flashing on your jostick. 
You will need the latest version of Development Center available on our website. Here is the link: 

https://www.kinovarobotics.com/en/knowledge-hub/all-kinova-products 
and choose "KINOVA® SDK GEN 2 Ultra lightweight robot 1.5.1".
Follow the release notes (RN) document to install the software properly.
Power-down the arm and short-circuit the 2 pins in the joystick connector as shown in the picture below. 
[image]

Connect the arm to the computer via USB, power-on the arm and launch Development Center.

·  In the top right corner, the mention "BOOTLOADER" should be displayed. 

·  Click on General Settings and go to the Update tab.

·  Click on Browse and choose the firmware version attached below for your mico 6DOF robot that you can download with this [link](https://drive.google.com/open?id=1DNkY1wlWMtOHItbMyxAOlY72H8vc5rV9) 

·  Click on Update (this step can take up to 2 minutes)

·  Once the update is completed, power down the arm and remove the short circuit from the joystick pins.

·  Power on the arm.

You should now be able to control the arm using the joystick. Now please make sure the developement center is closed and try to do the following two commands in your terminal   : 
1)roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s300
2)roslaunch m1n6s300_moveit_config m1n6s300_demo.launch

If there is still an error when you launch one of these two commands please let me know and I will help you :)

wangxian4423, 2019-04-27:
Ok,Thanks for your help.Now I'm waiting for the Google  Disks Premission.
[image]
And I'll try to update the firmware after that.Thanks!

wangxian4423, 2019-04-28:
Well,things seems to be difficult....I did what you said and it showed updated completed, howerer, the green lights still flash and things don`t seem to be better. So what can I do?

RalphChahine, 2019-04-29:
Hi wangxian,

could you please try to do the procedure again? I have tried it on my end on a mico that had the same problem you had and it worked. If it does not work after another attempt, it might be a hardware issue and you would have to consider sending the arm back at Kinova HQ for a fix.

Please let me know

wangxian4423, 2019-04-30:
Well,thanks for your help,I tried it again and the joystick still flash the green light,may there is something wrong with the joystick.And this time I try the kinova-ros-master devel and the moveit works well.I see the rqt_graph
[image]
and maybe there is a bug in the kinova-ros-kinectic devel which I used before as the rqt_graph in this version is:
[image]

g1y5x3, 2019-07-25:
Hi,
I had a similar issue with MICO. My Ubuntu was 16.04 and I was running ros kinetic. And the particular branch I was using was from this branch [kinetic devel](https://github.com/Kinovarobotics/kinova-ros/tree/kinetic)

What I realize that in the kinetic branch, it has its own robot controller written as a python script in **kinova_driver/src/scripts**, more specifically, they set the topic statically as "/j2s7s300/follow_joint_trajectory". This explains why when I was using MICO, the robot didn't move. It was simply because that driver was not configured for it. They also changed the launch file in **kinova_moveit** that calls the python script that they created.

To solve it, I simply replaced the [kinova_joint_trajectory_controller.cpp](https://raw.githubusercontent.com/Kinovarobotics/kinova-ros/master/kinova_driver/src/kinova_joint_trajectory_controller.cpp) from the master branch and **catkin_make** again. And my MICO arm works as expected!

Hope that would help!

Hubert51, 2019-08-08:
Can you give some more detail information? Currently, I only see `pid_traj_action_server.py` in kinova_driver/src/scripts. So what should I do to let my kinova arm execute the command? Do you mean I need to use `kinova_joint_trajectory_controller.cpp` from master to replace the same file in kinetic branch.
Thank you very much

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve