Kinovarobotics/kinova-ros #235: Sending multiple positions to the JACO2

Makekihe, 2019-04-26:
Hello, 
I'm using the JACO2SDK to program and control a JACO2 robotic arm on Ubuntu, C++. Here the purpose is to make the JACO2 move to a specific calculated cartesian position, while set to CARTESIAN_POSITION, close the grippers, and then move to another cartesian position. 

Currently I'm using the attached code to move the robot. This is only relevant parts of the code. When running the code, the robot will first initialise everything, and then afterwards move to a specified position in the grabObject function. Afterwards when trying to run the closeGripper function, the robot doesn't do anything. The program continues to call all the functions, and continues to do so in a loop, yet no movement is performed by the robot.

How is it possible to specify and send different commands (one after another) to the JACO2, using the JACO2SDK?
Additionally, do the JACO2 have a buffer in order to finish a motion, before calling the next one?

`//RETURN TO HOME POSITION
void returnToHome() {

	int(*MyGetCartesianCommand)(CartesianPosition &);
	MyGetCartesianCommand = (int(*)(CartesianPosition &)) dlsym(commandLayer_handle, "GetCartesianCommand");
	TrajectoryPoint pointToSend;
	pointToSend.InitStruct();
	pointToSend.Position.Type = CARTESIAN_POSITION;

	//We get the current position of the robot.
	MyGetCartesianCommand(currentCommand);

	//Setup coordinates for home position
	float homePositionX = 0.3f;
 	float homePositionY = 0.2f;
	float homePositionZ = 0.1f;

	cout << "Sending home" << endl;

	//Determine which direction the EE should move to the object in cartesian space
	pointToSend.Position.CartesianPosition.X = homePositionX;
	pointToSend.Position.CartesianPosition.Y = homePositionY;
	pointToSend.Position.CartesianPosition.Z = homePositionZ;
	pointToSend.Position.CartesianPosition.ThetaX = currentCommand.Coordinates.ThetaX;
	pointToSend.Position.CartesianPosition.ThetaY = currentCommand.Coordinates.ThetaY;
	pointToSend.Position.CartesianPosition.ThetaZ = currentCommand.Coordinates.ThetaZ;

	MySendBasicTrajectory(pointToSend2);
}

//CLOSE FINGERS
void closeFingers() {

	int(*MyGetCartesianCommand)(CartesianPosition &);
	MyGetCartesianCommand = (int(*)(CartesianPosition &)) dlsym(commandLayer_handle, "GetCartesianCommand");
	TrajectoryPoint pointToSend;
	pointToSend.InitStruct();
	pointToSend.Position.Type = CARTESIAN_POSITION;

	//We get the current position of the robot.
	MyGetCartesianCommand(currentCommand);

	cout << "Close fingers" << endl;

	//Determine which direction the EE should move to the object in cartesian space
	pointToSend.Position.CartesianPosition.X = currentCommand.Coordinates.X;
	pointToSend.Position.CartesianPosition.Y = currentCommand.Coordinates.Y;
	pointToSend.Position.CartesianPosition.Z = currentCommand.Coordinates.Z;
	pointToSend.Position.CartesianPosition.ThetaX = currentCommand.Coordinates.ThetaX;
	pointToSend.Position.CartesianPosition.ThetaY = currentCommand.Coordinates.ThetaY;
	pointToSend.Position.CartesianPosition.ThetaZ = currentCommand.Coordinates.ThetaZ;
	pointToSend.Position.Fingers.Finger1 = 6000;
	pointToSend.Position.Fingers.Finger2 = 6000;

	MySendBasicTrajectory(pointToSend1);
}

//MOVE TO AN OBJECT
void grabObject(vector <float> objectCoordinates) {

	float x = objectCoordinates[0];
	float y = objectCoordinates[1];
	float z = objectCoordinates[2];

	int(*MyGetCartesianCommand)(CartesianPosition &);
	MyGetCartesianCommand = (int(*)(CartesianPosition &)) dlsym(commandLayer_handle, "GetCartesianCommand");
	TrajectoryPoint pointToSend;
	pointToSend.InitStruct();
	pointToSend.Position.Type = CARTESIAN_POSITION;

	//We get the current position of the robot.
	MyGetCartesianCommand(currentCommand);

	cout << "grabObject" <<endl;

	//Determine which direction the EE should move to the object in cartesian space
	pointToSend.Position.CartesianPosition.X =  x;
	pointToSend.Position.CartesianPosition.Y =  y;
	pointToSend.Position.CartesianPosition.Z =  z;
	pointToSend.Position.CartesianPosition.ThetaX = currentCommand.Coordinates.ThetaX;
	pointToSend.Position.CartesianPosition.ThetaY = currentCommand.Coordinates.ThetaY;
	pointToSend.Position.CartesianPosition.ThetaZ = currentCommand.Coordinates.ThetaZ;

	//Send Trajectory
	MySendBasicTrajectory(pointToSend);

	closeFingers();
	returnToHome();

}
`

RalphChahine, 2019-04-26:
Hi, first of it is not good practice declare the API function pointers [int(*MyGetCartesianCommand)(CartesianPosition &);] and the function loaders [MyGetCartesianCommand = (int(*)(CartesianPosition &)) dlsym(commandLayer_handle, "GetCartesianCommand"); ] inside a function like you did it, you should make them available to your whole code. I suspect this might be the source of the problem, because I tried initializing the functions the same way you did and it did not work.

 To close the fingers here is the code I used and it worked very well: void closeFingers() {
	
	
	TrajectoryPoint pointToSend;
	CartesianPosition currentCommand;
	MyGetCartesianCommand(currentCommand);
	pointToSend.Position.Type = CARTESIAN_POSITION;

	pointToSend.Position.CartesianPosition.X = currentCommand.Coordinates.X;
	pointToSend.Position.CartesianPosition.Y = currentCommand.Coordinates.Y;
	pointToSend.Position.CartesianPosition.Z = currentCommand.Coordinates.Z;
	pointToSend.Position.CartesianPosition.ThetaX = currentCommand.Coordinates.ThetaX;
	pointToSend.Position.CartesianPosition.ThetaY = currentCommand.Coordinates.ThetaY;
	pointToSend.Position.CartesianPosition.ThetaZ = currentCommand.Coordinates.ThetaY;
	pointToSend.Position.Fingers.Finger1 = 6800;
	pointToSend.Position.Fingers.Finger2 = 6800;
	pointToSend.Position.Fingers.Finger3 = 6800;
	//We send the robot where we defined it to go
	MySendBasicTrajectory(pointToSend);

}

I also saw that in your close finger function, you initialize a trajectory point called pointToSend which has all your coordinates, but at the end of the function you send a basic trajectory with pointToSend1 which is not the point I think you are trying to send. 

Also the Jaco2 has indeed a buffer because it's main control loop is running at 100 Hz. Thus if you send lots of cartesian commands and or velocity commands it is crucial to have sleep times in order to ensure that the buffer does not get full so that you do not lose some commands when you perform a loop.

If you still have problems after reading this please send us a mail at support@kinova.ca and I will gladly do a live troubleshooting session with you.
Hope this helps :)

felixmaisonneuve, 2021-06-04:
I assume this answer has solve the issue, I will close it. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-04 by felixmaisonneuve