Kinovarobotics/kinova-ros #236: Joint Speed Limits mismatch between Code and Readme

smanschi, 2019-04-26:
There seems to be a mismatch between the joint speed limits in the Readme.md (10°/s for joints 1,2,3 and 20°/s for joints 4,5,6) and the actual limits used in the kinova_joint_angles_action.cpp file (20°/s for all joints).

What are the correct speed limits (for the gen2 jaco arm)? I need them as I am using the low-level C API for controlling the robots and have to limit the speed by myself.

nsaif-kinova, 2019-04-26:
Hello Smanschi,
By default the arm is configured to 10 deg/s.
Please refer to this file https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_bringup/launch/config/robot_parameters.yaml
You can go up to 48 deg/s on small actuators and 36 deg/s on the big ones.

smanschi, 2019-04-26:
Okay, thank you. I would recommend changing the values in the kinova_joint_angles_action.cpp file to 10 and 20 for consistency.

closed 2019-04-26 by smanschi