Kinovarobotics/kinova-ros #237: Arm stalks while giving demo in torque mode.

ZhengYi0310, 2019-04-26:
Hi,

When I put the arm in torque mode and move it with my own hands to give a demonstration, sometimes the arm will suddenly stalk, what might be the possible reason for this ? I already calibrated the torque sensor before putting the arm in torque mode and I moved the arm gentlly.

Another option I have is put the arm in admittance control mode and start doing demonstration, but the default force control parameters don't make the arm flexible enough for demonstration. And I think I might need to play around with different force control parameters using KinovaComm:setCartesianInertiaDamping and KinovaComm::kinova_comm_.setCartesianForceMinMax, espeically the damping coefficient. Is it ok if I just send a damping coefficient and leave other fields empty when using these two function ? And what would be a proper range of value ?

Thanks

nsaif-kinova, 2019-04-29:
Hello,
When you switch the arm to torque mode, there is a safety factor which will prevent the arm from moving at high and dangerous speed, it is called safety factor, you can set the safety factor to 1 so the arm will stop less, you can use the torque consol to do it or C++ SetTorqueSafetyFactor(1);

The function setCartesianInertiaDamping need the inertial parameters and can not be empty.
Sincerely,

ZhengYi0310, 2019-04-29:
It's already been set to 1, I did that everytime before I set the arm to torque mode for demonstration, yet the arm still stalls. Should I try a lower number?

nsaif-kinova, 2019-04-29:
Hello,
If you use lower values the arm will have a higher safety factor and will stop more often.
Is your arm going into a self collision?
Did you calibrate your torque to zero? set the gravity vector appropriately?
Can you make sure the zero position of your actuators is correct? please refer to page 46 Position indexation
https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf

ZhengYi0310, 2019-04-29:
I always put the arm to candle position [180, 180, 180, 180, 180, 180, 180], and call ZeroTorque(), then move the arm back to the start position I set, and then put it in the torque mode and start demonstration. For the gravity vector, I just use the default one set in the package because the arm is in a standard vertical position. I don't think the arm is going to self collision, because my demo is actually causing joints move away from each other, but I can post pics of my set up so you can have a more professional jugement.

ZhengYi0310, 2019-04-29:
This is what I'm doing, I launch the driver of the arm at the following start position:
[image]

Then, I commanded the arm to move to the handle position:
[image]

Than I called KinovaComm::setZeroTorque() to calibrate the torque sensor. Then I moved the arm to the home position again, wait for several seconds, and called for KinovaComm::setToquesControlSafetyFactor(1) and KinovaComm::SetTorqueControlState(1), finally I called float toques[7] = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0};KinovaComm::JointTorques(toques) to put the arm into "gravity compensation mode" so I can start doing demonstration.

I did a lot of tests, sometimes, the arm will just stall at the start position, and my suspect is that this start position might still trigger the self-collision avoidance mechanism, but I'm not sure since I don't know the detail of the code for self-collision. If this is the case, is there anyway I can disable this feature using the api ? Since I'm not actually doing any torque control and just trying to give a demonstration.

Sometimes, the arm will also stall at a configuration after some movements, like the following:
[image]. Although I think I already moved the arm quite gently, the torque I applied might still be larger than the default maximum joint torque, since I don't know what is the default maximum torque for joints. If this is the case, I guess I need to use KinovaComm::setJointTorqueMinMax(AngularInfo &min, AngularInfo &max) to reset the torque limits?

Thanks

nsaif-kinova, 2019-04-29:
Hello,
Please send me an email to support@kinova.ca with your availability for a conference call, it will be easier and faster to discuss your issue over the phone.
Sincerely,

nsaif-kinova, 2019-04-29:
Please also note the torque will stop before the self collision happens to prevent it, please refer to the angular limit page 44 https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf
Are you able to switch to cartesian mode while the arm is in the position shown in the first picture?I think your 4th joint is at its limit.

ZhengYi0310, 2019-04-29:
Hi Nassif,
    Thanks very much for the information ! Yeah I think my start position for joint 4 being close to its limit is very much likely the potential problem, is there some way to set a "safety margin" for self collision, just like the safety factor for torque ? Also, for the KinovaComm::setJointTorqueMinMax(AngularInfo &min, AngularInfo &max), is the minimum value here the absolute value of the torque or the sign of it should also be considered ?

    Thanks

ZhengYi0310, 2019-04-30:
Hi Nassif,
    I also tried the following start position, where the joint 4 is decently far from its limit:
    
[image]

    But under this configuration the situation seems get worse, the arm either stalls at the start of my demonstration, or shortly after I start my demonstration.

   Thanks

nsaif-kinova, 2019-04-30:
Hello Yi,
Unfortunately you can not set a safety zone while in torque mode, but you can do it in Cartesian mode.
It is better to always send the arm to home position before you switch to torque mode.
Please refer to this question "My robot will not switch from position mode to torque mode" page 65 https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf
Sincerely,

closed 2019-05-02 by ZhengYi0310