Kinovarobotics/kinova-ros #238: Too high velocity in trajactory execution (melodic)

Julia794, 2019-04-30:
I am working with Mico on ros melodic, ubuntu 18.08. 
Moveit (or OMPL) seems to set velocities, that are higher than needed. This causes wrong endpositions. 
I tried to execute a rotation of link 6 of 50 degree (from 55.7 to 105.7 degree and back again) but it moves around 5 to 6 degree too far (50.5 and 111.5). 

The follow_joint_trajectory publishes correct position, but velocity completely zero. trajectory_controller/command publishes  correct position too, but the velocity which seems to bee to high, which is also too high in ../in/joint_velocity and trajectory_command/state. 

Hope someone could help. I will try to restructure my code so i can publish it in here for more information.

alexvannobel, 2019-04-30:
Hi @Julia794 ,

The Melodic support for this repo is a work in progress currently, but I remember issue #200 where we found out the velocity trajectory controller could produce some overshoot situations in intermediate and final positions.

Maybe the changes we made, that currently live in [this branch](https://github.com/Kinovarobotics/kinova-ros/tree/dev/reduce-overshoot-follow-joint-trajectory) could help you. 
Also, have you tried modifying the MoveIt soft limits for velocity and acceleration? You might be able to reduce the trajectories' velocities if you set them lower.

Hope this helps,

Cheers,
Alex

christian-rauch, 2019-05-13:
@Julia794 Did you try the melodic changes in https://github.com/Kinovarobotics/kinova-ros/pull/219 ?
With this, I was able to set the end-effector pose via the MoveIt plugin in RViz.

Julia794, 2019-05-28:
Thanks for the answers. Unfortunatly none of these solutions work. 
If I give a Jointpose which only moves one Joint to a defined position it overshoots a lot. Planned movement from 57 degree to 107 degree and back again reaches Positions around 111.5 and 52.5 degrees. 
I'm trying to find out where the problem comes from. 
The given velocity is executed correctly by the arm.  So it might be anything in the planned path by moveit. Maybe caused by some wrong settings? 
I'm about to test if a given trajectory is executed right.

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over a years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve