Kinovarobotics/kinova-ros #239: Arm found independently of type

Kaysera, 2019-05-02:
Good afternoon. 

After connecting our arm via USB (Kinova Mico 2, 6DOF 3 fingers), and using the bring_up instruction, ROS will say the robot is detected, whether I write the robotType for Jaco or Mico. However, none of both will make the robot open the fingers (it says fingers should be opening before start), even though ROS seems to complete the execution. It seems ROS is not publishing in the proper topic due to not having the proper model configured. Any idea on how to detect the model, or see the topic where the robot is reading?

nsaif-kinova, 2019-05-02:
Hello Guillermo,
What is the type of the arm? Jaco or mico? 6 or 7 dof? curved or spherical wrist?
What are the commands you are using?
Did you initialize the fingers before you start your ROS?
Can you use the arm with the Dev Center?
Sincerely,

Kaysera, 2019-05-09:
Hello nsaif. Thanks for your response. The problem was that I did not copy the udev rule file. I'm sorry for the inconvenience. 

Cheers.

closed 2019-05-09 by Kaysera