Kinovarobotics/kinova-ros #241: Joint effort and velocity limits in kinova_description

ferrolho, 2019-05-09:
I think the values in [`kinova_description/urdf/kinova_common.xacro`](https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro) for the joint _effort_ and _velocity_ limits are not set correctly. Are they really `2000` and `1`?

What are the true limits or in which spec sheet are they specified? Thanks!

https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L61

https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L89

https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L104

https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L126

christian-rauch, 2019-08-28:
The finger joint limits:
https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L104
https://github.com/Kinovarobotics/kinova-ros/blob/73ef56dca89400e6f19e07753d241e900cc78aca/kinova_description/urdf/kinova_common.xacro#L126
also appear arbitrary. The upper bound of 2 radians for the proximal finger is physically implausible.

@alexvannobel Can you check if these values are correct?

alexvannobel, 2019-09-05:
Hi guys, 
Sorry for the long delay.
Definitely those values are not good. I'll look quickly in the specs sheet to have more plausible values (for actuators and fingers).
I'll keep you updated soon.

Best,
Alex

alexvannobel, 2019-09-09:
Hi guys,

I found velocity limits values in the [specs sheet for Jaco2](https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf).
I found torque limits in the base's firmware which should be more realistic.
I also added a more realistic upper limit for the finger's position based on experimental testing with a Jaco2 here. The velocity limit was alright. I'm waiting on an answer for the finger's effort limit.  

I'll keep you up-to-date whenever the PR gets merged. 
Best,
Alex

alexvannobel, 2019-09-10:
Hi guys,
 
#266 was just merged for joint limits for both actuators and fingers.

Feel free to reopen this issue if you experience problems with the new values.
Best,
Alex

closed 2019-09-10 by alexvannobel

ferrolho, 2019-09-10:
```zsh
➜  urdf git:(master) ✗ rosrun xacro xacro -o j2n6s200_standalone.urdf j2n6s200_standalone.xacro
Undefined parameters [joint_velocity_limit,joint_torque_limit]
when instantiating macro: kinova_armjoint (/home/henrique/catkin_ws/src/kinova-ros/kinova_description/urdf/kinova_common.xacro)
instantiated from: j2n6s200 (/home/henrique/catkin_ws/src/kinova-ros/kinova_description/urdf/j2n6s200.xacro)
in file: j2n6s200_standalone.xacro
```