Kinovarobotics/kinova-ros #243: What is the joint angle resolution from the jaco2？

bzdfzfer, 2019-05-18:
May I ask the encoder resolution value in the Jaco2 robot arm?

nsaif-kinova, 2019-06-13:
Hello,
I have the position sensor reolution for Ka-75+ 3,686,400/turn and for Ka-58 2,534,400/turn
Please refer to this document : https://www.kinovarobotics.com/sites/default/files/AS-ACT-KA58-KA75-SP-INT-EN%20201804-1.2%20%28KINOVA%E2%84%A2%20Actuator%20series%20KA75%2B%20KA-58%20Specifications%29.pdf
Let me know if this answers your question, if you have other hardware questions I invite you to send an email to supoprt@kinova.ca

bzdfzfer, 2019-06-13:
> Hello,
> I have the position sensor reolution for Ka-75+ 3,686,400/turn and for Ka-58 2,534,400/turn
> Please refer to this document : https://www.kinovarobotics.com/sites/default/files/AS-ACT-KA58-KA75-SP-INT-EN%20201804-1.2%20%28KINOVA%E2%84%A2%20Actuator%20series%20KA75%2B%20KA-58%20Specifications%29.pdf
> Let me know if this answers your question, if you have other hardware questions I invite you to send an email to [supoprt@kinova.ca](mailto:supoprt@kinova.ca)

Yes! The document helps a lot! Thanks a lot for your reply！

closed 2019-06-13 by bzdfzfer