Kinovarobotics/kinova-ros #244: Flashing Red Light on Jaco Arm

ericrosenbrown, 2019-05-30:
## Description

We currently have our jaco arm mounted on a table, and are trying to interface with it. When we plug in the controller, we first get a few green lights flashing, and then after a bit we see red lights start to flash. According the guide, the reasons the lights would flash red are for the following reasons:

## Version

ROS distribution : 

Branch and commit you are using : 

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

nsaif-kinova, 2019-06-13:
Hello Eric,
A red flashing light indicate a hardware or firmware issue, please contact us at support@kinova.ca, and we will help you solve this problem quickly.
Sincerely,

polthergeist, 2020-03-20:
Since there is no activity in this issue, I will close it. Feel free to reopen it if the issue was not solved.

closed 2020-03-20 by polthergeist

burakaksoy, 2022-08-23:
I have the same issue. Would you please share what was the problem in your case and how did you solve? I also emailed to Kinova support but I am waiting for their reply..