Kinovarobotics/kinova-ros #245: Mico arm won't move after a day

kevincrawfordknight, 2019-05-31:
## Description

After a day (or so) of idle time, the robot arm stops responding to move commands.  (The fingers still work, though!)  I never used to have this problem -- the robot is expected to stay working for several days.

Error:

[ WARN] [1559344236.439846379]: void kinova:KinovaAnglesActionServer::actionCallback(const ArmJointAngelsGoalConstPtr&): LINE 161, Trajectory command failed

## Version

ROS distribution : kinetic

Branch and commit you are using :   5.2.0

## Steps to reproduce

1. Turn on arm.
2. Launch ros
3. Issue command below (robot moves)
4. Wait a day
5. Issue same command (robot doesn't move)

## Code example (if necessary)

rosrun kinova_demo joints_action_client.py -v -r m1n4s200 degree -- 10 10 10 10

## Expected behavior

Robot should still work after idle time.

nsaif-kinova, 2019-06-13:
Hello Kevin,
Can you use a joystick to control the arm before a reboot?
Can you operate the arm normally after a reboot using ROS?
If you are unable to operate the arm using a joystick before you reboot it or if the joystick is flashing red or not responding this might not be related to ROS, and I invite you send an email to support@kinova.ca
Sincerely,

polthergeist, 2020-03-20:
Hi  @kevincrawfordknight ,
I was wondering, is this issue resolved?

closed 2021-06-04 by felixmaisonneuve