Kinovarobotics/kinova-ros #247: Issue with MoveIt on Kinetic

gadese, 2019-06-12:
## Description

I recently moved from Ubuntu 14.04 with ROS Indigo to Ubuntu 16.04 with ROS Kinetic. It's a brand new machine.

I'm trying to control the Mico arm (m1n6s300) in Gazebo using MoveIt!. It worked fine on my old Indigo version but something seems to be wrong with my new setup because MoveIt! keeps giving me the following error:

```
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'm1n6s300_joint_2': expected: 2.90611, current: 2.89302"

ABORTED: Solution found but controller failed during execution
```
[image]

From what I understand, it means that the start point is different than what is actually expected. However, I'm not sure why.

My guess is an issue with the ros_controllers (or a version issue?), as the robot is visibly shaking/wobbly in Gazebo, even when no command is given and MoveIt isn't running.

Another hint that the controllers may be an issue is that about 30% of the time, when I spawn the arm in Gazebo, it fails with the following error code, with the robot seemingly having a seizure on the ground:

```
Unexpected exception caught when initializing trajectory from ROS message data.
```
[image]

## Version

ROS distribution : Kinetic

Branch and commit you are using : 
kinova-ros
branch "kinetic"
latest commit (eed3c528dfd5a7e2f5cd1871023da89904518ff0)

gazebo_ros version: 2.5.18
gazebo_ros_control version: 2.5.18
moveit version: 0.9.15

## Steps to reproduce

1. Launch Gazebo with 

```
roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=m1n6s300
```
This is where the "Unexpected exception..." error sometimes pops up

2. Launch the MoveIt RViz plugin with

```
roslaunch m1n6s300_moveit_config m1n6s300_gazebo_demo.launch
```
This correctly boots RViZ, with the apparently harmless warnings:

```
Skipping virtual joint 'world_frame_fixed_joint' because its child frame 'root' does not match the URDF frame 'world'
```
```
Interactive marker 'EE:goal_m1n6s300_end_effector' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.
```

3. Try to move the arm in RViZ. I used the pick_and_place script back in Indigo, but here I'm just trying to go back to the basics so I use the interactive markers in RViZ to move the arm to a different position, then click "Plan and Execute". This is where it crashes with the "Invalid Trajectory" error shows up.

## Expected behavior

When I used Indigo, those same commands seemed to work without issue.

gadese, 2019-07-10:
Hi, I was wondering if there was any follow up? @KinovaCustomerSupport

alexvannobel, 2019-07-16:
Hi @gadese ,
Sorry for the delay.

I just merged a PR to master which should fix this in ROS Kinetic.
Feel free to reopen the issue if you encounter any issue due to this merge.

Best,
Alex

closed 2019-07-16 by alexvannobel