Kinovarobotics/kinova-ros #248: Flashing Green Light on Jaco Arm

Hubert51, 2019-06-26:
We currently have our jaco arm mounted on a table, and are trying to interface with it. When we plug in the controller, we get green lights flashing all the time for at least 20 minutes. Does something wrong? Thank you for helping!

Hubert51, 2019-06-26:
I ask the support in kinova. The problem is the firmware is out of date. We need to update the firmware of the arm. The following is the step instruction to update the firmware

1. First of all, you need to download the latest firmware version of your arm. You can download it here. Choose the one suited for your arm type (Spherical or standard wrist) ) : https://www.kinovarobotics.com/en/knowledge-hub/gen2-ultra-lightweight-robot

2. You will also need the latest version of Development Center available here : https://www.kinovarobotics.com/en/knowledge-hub/all-kinova-products and choose "KINOVA® SDK GEN 2 Ultra lightweight robot 1.5.1". Follow the release notes (RN) document to install the software properly.

3. On your arm:  Power-down the arm and short-circuit the 2 pins in the joystick connector as shown in the picture below.
(**I do not do this step**)

4. Connect the arm to the computer via USB, power-on the arm and launch Development Center. 
* In the top right corner, the mention "BOOTLOADER" should be displayed.
* Click on General Settings and go to the Update tab.
* Click on Browse and choose the firmware version downloaded previously.
* Click on Update (this step can take up to 2 minutes)
* Once the update is completed, power down the arm and remove the short circuit from the joystick pins.
* Power on the arm.

closed 2019-06-26 by Hubert51

bradsa, 2022-12-26:
Thanks for this!