Kinovarobotics/kinova-ros #249: j2n6s300 in Gazebo the model spawn not correctly

rachelhson, 2019-07-04:
I launched  Gazebo with 
 
roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2n6s300

The model shown on simulation starts bouncing and then each part of the arm stuck in the ground. 

Could you please provide any advice for this issue?  

Thank you.

kotzir, 2019-07-08:
I was facing the same issues, and used this package for the simulations:

[https://github.com/JenniferBuehler/jaco-arm-pkgs/wiki/Installation](url)

alexvannobel, 2019-07-16:
Hi @rachelhson ,

I just merged a PR which should fix this bouncing issue. I tested it on ROS Kinetic with Gazebo 7 and the robot behaved properly.

Feel free to reopen this issue if you encounter any problem due to this merge.

Best,

Alex

closed 2019-07-16 by alexvannobel