Kinovarobotics/kinova-ros #251: robot_state_publisher: symbol lookup errorliburdf.so: undefined symbol

g1y5x3, 2019-07-19:
## Description
When launching **roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s200**, receives the following error

/opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /opt/ros/kinetic/lib/liburdf.so: undefined symbol: _ZN4urdf9parseURDFERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

## Version
Ubuntu 16.04
ROS distribution : 
kinetic
Branch and commit you are using : 
kinetic

## Steps to reproduce
1. roslaunch kinova_bringup kinova_robot.launch

## Any other information
The robot driver still works properly, I can still see all the messages published onto the m1n6s200_driver/* topic, however, when I echo the /tf topic, nothing is showing.

alexvannobel, 2019-07-22:
Hi @g1y5x3 ,

This looks more like a missing/wrong dependancies problem than a kinova-ros problem.
Maybe [this post](https://answers.ros.org/question/282495/rviz-symbol-lookup-error/) could help you resolve your issue.
Did you try uninstalling ROS and then reinstalling it? Can you launch another robot in RViz without getting this error?

Best,
Alex

g1y5x3, 2019-07-22:
Thanks!
That thread was very helpful and it was the dependancies issue.
When I did 
`ldd /opt/ros/kinetic/lib/liburdf.so | grep urdf`
I got the following
`liburdfdom_model.so.0.4 => /usr/local/lib/liburdfdom_model.so.0.4 (0x00007f4328dca000)`

However, if I tried to locate the the file
`locate liburdfdom_model`
It shows that it was in a different directory
`/usr/lib/x86_64-linux-gnu/liburdfdom_model.so`
`/usr/lib/x86_64-linux-gnu/liburdfdom_model.so.0.4`
`/usr/lib/x86_64-linux-gnu/liburdfdom_model_state.so`
`/usr/lib/x86_64-linux-gnu/liburdfdom_model_state.so.0.4`

What I did was the following two instructions:
`export LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/:$LD_LIBRARY_PATH `
`export LD_LIBRARY_PATH=/opt/ros/kinetic/lib:$LD_LIBRARY_PATH`

The second one was to make sure that `liburdf.so` is still using the one from ros not in my /usr/lib/

closed 2019-07-23 by alexvannobel