Kinovarobotics/kinova-ros #252: launch two arms driver in the same time

ZakariaChekakta, 2019-07-26:
## Summary
We have two arms of kinova JACO2, we are going to use them both with mobile robot base, however, when we launch the drivers for each robot separately they are working well, but when we put all the driver in the same launch file and launch them in the same time, one arm launch well, and  we got an error in the second one, the driver couldn't reach to the arm using the serial number, I tried this in both connection weather USB or Ethernet,  

## Use case
Two Jaco 2 arm with mobil base robot

## Alternative

For now, when I am launching both in the same file , I just delayed the second one 5 second, and it is works well, but it is not an efficient solution

alexvannobel, 2019-08-08:
Hi @ZakariaChekakta ,
Can you provide a sample of your launch file? I'll try to reproduce your launch on my side. 
Did the delay do the trick for both the USB and Ethernet connection or was there a connection type that did not work with 2 arms? For Ethernet, you will have to specify different port numbers for each driver for the API to initialize correctly.

ZakariaChekakta, 2019-08-13:
[image]

This is my launch file 
@alexvannobel , sorry I didn't reply so fast

closed 2021-06-04 by felixmaisonneuve